SpikeInterface/spikeinterface #209: Bug: Multisorting - get_unit_spike_train

marcbue, 2021-07-30:
I created a multisorting following the tutorial for spikeinterface 0.90. After creating my multisorting I can extract all spike trains using multisorting.get_all_spike_trains(). But I can't extract it for a single unit using multisorting.get_unit_spike_train(0).

The following error occurs:

---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-24-88b592715d21> in <module>
----> 1 multisorting.get_unit_spike_train(0)

~/miniconda3/envs/spikesorting/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/core/basesorting.py in get_unit_spike_train(self, unit_id, segment_index, start_frame, end_frame)
     54                              end_frame: Union[int, None] = None,
     55                              ):
---> 56         segment_index = self._check_segment_index(segment_index)
     57         S = self._sorting_segments[segment_index]
     58         return S.get_unit_spike_train(unit_id=unit_id, start_frame=start_frame, end_frame=end_frame)

~/miniconda3/envs/spikesorting/lib/python3.8/site-packages/spikeinterface-0.90.0.dev0-py3.8.egg/spikeinterface/core/base.py in _check_segment_index(self, segment_index)
     63                 return 0
     64             else:
---> 65                 raise ValueError()
     66         else:
     67             return segment_index

ValueError: 

---------------------------------------------------------------------------

Because of this many functions do not work with the multisorting as I would expect. For example saving it to npz or using it for plotting. (se.NpzSortingExtractor.write_sorting(multisorting, save_path=str(output_folder / 'sorting.npz')))

Also is there a way to extract individual sortings from the segments of a multisorting?

alejoe91, 2021-07-30:
Hi,

How did you create exactly the multisorting object?

alejoe91, 2021-07-30:
Can you try to reinstall from master? I think I know where the error is and we fixed it. 

In your case (before the fix), you should be able to get single spike trains by specifying the `segment_index`:
```
spike_train0 = multisorting.get_unit_spike_train(0, segment_index=0)
```

marcbue, 2021-07-31:
`spike_train0 = multisorting.get_unit_spike_train(0, segment_index=0)`

This works!

I still have problems if functions rely on the get_unit_spike_train() function.

E.g.

`sorting_cached = se.CacheSortingExtractor(multisorting, 'sorting_cache.npz')`

Or also the plots that you could use with the sorting.

Also a question. If eveything works as intended. Is it possible to use the multisorting in a similar fashion as a sorting? E.g. for using widgets and plots that exist? If not in that case the question if one could extract a usual "single" sorting from a multisorting segment to plot for example rasterplots (and other more complex beautiful stuff you implemented) only for this segment.

alejoe91, 2021-08-01:
Hi @marcbue 

Sure the `multisorting` is a `SortingExtractor` object, so everything should work :) 

One question: is your `multisorting` object an agreement sorting (after multicomparison) or is it 'just' a multi-segment object?

We just released a new version for the new API (0.90.0). If you have any problems, don't hesitate to open an issue. We are sure there are still bugs to fix!

marcbue, 2021-08-02:
Hi @alejoe91 

not sure how to answer your question. This is what I get for checking the type:

[image]

The get_unit_spike_train issue still prevails for me for these multirecordings

One question about the new release:
I uninstalled all the old packages now and install 0.90 via the pip repo. 
Loading recordings that I processed before with an older version e.g with this code:

```
filtered_file_df = file_df_filtered_out_faulty[file_df_filtered_out_faulty.experiment==file_df_filtered_out_faulty.experiment.unique()[55]].sort_values('file_code', axis=0)
recording_files = filtered_file_df.destination_path.to_list()
intermediate_folders = filtered_file_df.destination_dir.apply(lambda x: x / 'intermediate').to_list()
output_folder = filtered_file_df.destination_path.iloc[0].parents[3] / 'merged' / 'merged'
recordings = []
for (rec_path, inter_path) in zip(recording_files, intermediate_folders):
    cache_folder = inter_path / 'cache' / sorter
    rec_cache = si.load_extractor(cache_folder.parent / 'recording.pkl')
    rec = si.BinaryRecordingExtractor(file_paths=cache_folder.parent / 'recording.dat', 
                                          sampling_frequency=rec_cache.get_sampling_frequency(), 
                                          num_chan=rec_cache.get_num_channels(), 
                                          dtype='float32', 
                                          channel_ids=rec_cache.get_channel_ids(), 
                                          gain_to_uV=rec_cache.get_channel_gains()
                                         )
    rec.set_channel_locations(rec_cache.get_channel_locations())
    recordings.append(rec_cache)
multirecording = si.append_recordings(recordings)
probe = Probe(ndim=2, si_units='um')
probe.set_contacts(positions=rec_cache.get_channel_locations(), shapes='circle')
probe.set_device_channel_indices(np.arange(rec_cache.get_num_channels()))
multirecording = multirecording.set_probe(probe)
multisorting = si.run_sorter(sorter, multirecording, output_folder=output_folder, verbose=True, remove_existing_folder=True)
```

Throws now:

```
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
<ipython-input-23-29e4de58322f> in <module>
      6 for (rec_path, inter_path) in zip(recording_files, intermediate_folders):
      7     cache_folder = inter_path / 'cache' / sorter
----> 8     rec_cache = si.load_extractor(cache_folder.parent / 'recording.pkl')
      9     rec = si.BinaryRecordingExtractor(file_paths=cache_folder.parent / 'recording.dat', 
     10                                           sampling_frequency=rec_cache.get_sampling_frequency(),

~/miniconda3/envs/spikesorting/lib/python3.8/site-packages/spikeinterface/core/base.py in load_extractor(file_or_folder_or_dict, base_folder)
    758         return BaseExtractor.from_dict(file_or_folder_or_dict, base_folder=base_folder)
    759     else:
--> 760         return BaseExtractor.load(file_or_folder_or_dict, base_folder=base_folder)
    761 
    762 

~/miniconda3/envs/spikesorting/lib/python3.8/site-packages/spikeinterface/core/base.py in load(file_path, base_folder)
    472             else:
    473                 raise ValueError(f'Impossible to load {file_path}')
--> 474             extractor = BaseExtractor.from_dict(d, base_folder=base_folder)
    475             return extractor
    476 

~/miniconda3/envs/spikesorting/lib/python3.8/site-packages/spikeinterface/core/base.py in from_dict(d, base_folder)
    298             The loaded extractor object
    299         '''
--> 300         if d['relative_paths']:
    301             assert base_folder is not None, 'When  relative_paths=True, need to provide base_folder'
    302             d = _make_paths_absolute(d, base_folder)

KeyError: 'relative_paths'
```

Do I have to redo the old recordings now with the new version?

alejoe91, 2021-08-02:
Hi @marcbue 

sorry for the confusion. Yes the old cache mechanism has been improved. You can now use the `save()` function directly (note that now writing these binary files in parallel is much faster as we improved the parallel processing backend). 
Sorry about the inconvenient, but I strongly recommend you re-save your recording using the new version.

alejoe91, 2021-08-02:
BTW the error with `get_unit_spike_train` is not an actual error, but an expected behavior. Since you have more segments, you have to specify which segments you want spike trains from. If there is only one segment, than you can aviod specifying the segment index.

marcbue, 2021-08-14:
Thanks Alessio! Finally came around to test the data now fully in the new spikeinterface and it works great! Both saving and get_unit_spike_train.

closed 2021-08-14 by marcbue