SpikeInterface/spikeinterface #216: Question: MCSH5RecordingExtractor

marcbue, 2021-08-09:
Hi,

I want to create a recording for the MCS system. Before the new release I used MCSH5RecordingExtractor which seemed to work reasonable well. I saw that this is not implemented anymore (if I'm not mistaken).
What would be the best way to create a recording in this case? (I saw that  MCSRawRecordingExtractor exists but the data is in the h5 format unfortunately).
Should I create a numpyrecording or create an extractor based on BaseRecording or what would be the best way to implement this?

Thanks!

marcbue, 2021-08-14:
I've managed to do it creating manually a NumpyRecording after extracting necessary infos from the H5 file. So I'll close this.

closed 2021-08-14 by marcbue

samuelgarcia, 2021-08-23:
Note that MCS data can be also exported to  RAW+MCS header and so you can use instead  the `MCSRawRecordingExtractor` or  `read_mcsraw()`