SpikeInterface/spikeinterface #221: Auto- and cross-correlation

BenATemple, 2021-08-13:
I am trying to plot auto and crosscorrelograms.
I notice that the tutorial for these plots (as well as ISI distribution) is not complete.
I try this: sw.plot_autocorrelograms(sorting_IC, unit_ids=[3], window_ms=100.0, bin_ms=1.0, symmetrize=False)
But I get an error : [AssertionError: 'axes' can be a 1-d array or list or a 2d array of axis]
Seems like an easy fix but not sure exactly what to do.
AC, CC, and ISI are nice quality-metric visualizations so would be helpful to flesh this out if anyone has tips.
Thank you,
Ben

alejoe91, 2021-08-17:
Hi @BenATemple 

Seems like a small bug on our side. I'll look into it next week :) 
No sfn for us this year, but I'm giving a SI tutorial next Friday for the nwb user days workshop: https://www.incf.org/neurodata-without-borders-nwb-user-training-workshop

BenATemple, 2021-08-17:
Thanks a lot, @alejoe91 ! And I'll send the link to the tutorial to some others who might be interested :)

samuelgarcia, 2021-08-23:
Is this fixed ?
Can you copy/paste a short example that trigger the bug ?

BenATemple, 2021-08-23:
Thanks, Samuel!
To my knowledge, it its not yet fixed. I just attempted again.
I try this: sw.plot_autocorrelograms(sorting_IC, unit_ids=[3], window_ms=100.0, bin_ms=1.0, symmetrize=False)
But I get an error : [AssertionError: 'axes' can be a 1-d array or list or a 2d array of axis]

samuelgarcia, 2021-08-24:
See #230 
I have updated examples for auto and cross correlogram.
Example are working on my side.

samuelgarcia, 2021-09-17:
@BenATemple :  can you check that examples are working for you ?

BenATemple, 2021-09-21:
@samuelgarcia Sorry, just got back to this. Yes, it works now, thank you!

closed 2021-09-22 by samuelgarcia