SpikeInterface/spikeinterface #223: Using Dockers

BenATemple, 2021-08-15:
Hello,
I'm trying to use dockers (which I don't have experience with) to run some of the spike sorters (which are, indeed, sometimes difficult to install).
I follow your instructions: to (1) have docker installed and (2) to have docker python SDK installed (pip install docker)
However, when I try to run a sorter (MountainSort4) I run into errors (see below).
Any quick tips for how to make this work?
Thank you,
Ben

**Command in Jupyter Notebook**: sorting_MS = ss.run_mountainsort4(recording_preprocessed, output_folder='results_Tetrode', docker_image = "spikeinterface/mountainsort4-base")

**Error Message in Jupyter Notebook**
HTTPError                                 Traceback (most recent call last)
~\anaconda3\lib\site-packages\docker\api\client.py in _raise_for_status(self, response)
    267         try:
--> 268             response.raise_for_status()
    269         except requests.exceptions.HTTPError as e:

~\anaconda3\lib\site-packages\requests\models.py in raise_for_status(self)
    942         if http_error_msg:
--> 943             raise HTTPError(http_error_msg, response=self)
    944 

HTTPError: 404 Client Error: Not Found for url: http+docker://localnpipe/v1.41/containers/create

During handling of the above exception, another exception occurred:

ImageNotFound                             Traceback (most recent call last)
~\anaconda3\lib\site-packages\docker\models\containers.py in run(self, image, command, stdout, stderr, remove, **kwargs)
    810         try:
--> 811             container = self.create(image=image, command=command,
    812                                     detach=detach, **kwargs)

~\anaconda3\lib\site-packages\docker\models\containers.py in create(self, image, command, **kwargs)
    869         create_kwargs = _create_container_args(kwargs)
--> 870         resp = self.client.api.create_container(**create_kwargs)
    871         return self.get(resp['Id'])

~\anaconda3\lib\site-packages\docker\api\container.py in create_container(self, image, command, hostname, user, detach, stdin_open, tty, ports, environment, volumes, network_disabled, name, entrypoint, working_dir, domainname, host_config, mac_address, labels, stop_signal, networking_config, healthcheck, stop_timeout, runtime, use_config_proxy)
    429         )
--> 430         return self.create_container_from_config(config, name)
    431 

~\anaconda3\lib\site-packages\docker\api\container.py in create_container_from_config(self, config, name)
    440         res = self._post_json(u, data=config, params=params)
--> 441         return self._result(res, True)
    442 

~\anaconda3\lib\site-packages\docker\api\client.py in _result(self, response, json, binary)
    273         assert not (json and binary)
--> 274         self._raise_for_status(response)
    275 

~\anaconda3\lib\site-packages\docker\api\client.py in _raise_for_status(self, response)
    269         except requests.exceptions.HTTPError as e:
--> 270             raise create_api_error_from_http_exception(e)
    271 

~\anaconda3\lib\site-packages\docker\errors.py in create_api_error_from_http_exception(e)
     30             cls = NotFound
---> 31     raise cls(e, response=response, explanation=explanation)
     32 

ImageNotFound: 404 Client Error for http+docker://localnpipe/v1.41/containers/create: Not Found ("No such image: spikeinterface/mountainsort4-base:latest")

During handling of the above exception, another exception occurred:

HTTPError                                 Traceback (most recent call last)
~\anaconda3\lib\site-packages\docker\api\client.py in _raise_for_status(self, response)
    267         try:
--> 268             response.raise_for_status()
    269         except requests.exceptions.HTTPError as e:

~\anaconda3\lib\site-packages\requests\models.py in raise_for_status(self)
    942         if http_error_msg:
--> 943             raise HTTPError(http_error_msg, response=self)
    944 

HTTPError: 404 Client Error: Not Found for url: http+docker://localnpipe/v1.41/images/create?tag=latest&fromImage=spikeinterface%2Fmountainsort4-base

During handling of the above exception, another exception occurred:

NotFound                                  Traceback (most recent call last)
<ipython-input-30-1a7d7a0333f1> in <module>
----> 1 sorting_MS = ss.run_mountainsort4(recording_preprocessed, output_folder='results_Tetrode', docker_image = "spikeinterface/mountainsort4-base")

~\anaconda3\lib\site-packages\spikeinterface-0.90.0.dev0-py3.8.egg\spikeinterface\sorters\runsorter.py in run_mountainsort4(*args, **kwargs)
    250 
    251 def run_mountainsort4(*args, **kwargs):
--> 252     return run_sorter('mountainsort4', *args, **kwargs)
    253 run_mountainsort4.__doc__ = _common_run_doc.format('mountainsort4')
    254 

~\anaconda3\lib\site-packages\spikeinterface-0.90.0.dev0-py3.8.egg\spikeinterface\sorters\runsorter.py in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, **sorter_params)
     55             verbose=verbose, raise_error=raise_error, **sorter_params)
     56     else:
---> 57         sorting = run_sorter_docker(sorter_name, recording, docker_image, output_folder=output_folder,
     58                 remove_existing_folder=remove_existing_folder, delete_output_folder=delete_output_folder,
     59                 verbose=verbose, raise_error=raise_error, **sorter_params)

~\anaconda3\lib\site-packages\spikeinterface-0.90.0.dev0-py3.8.egg\spikeinterface\sorters\runsorter.py in run_sorter_docker(sorter_name, recording, docker_image, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, **sorter_params)
    171     cmd = 'python -c "import spikeinterface; print(spikeinterface.__version__)"'
    172     try:
--> 173         res = client.containers.run(docker_image, cmd)
    174         need_si_install = False
    175     except docker.errors.ContainerError:

~\anaconda3\lib\site-packages\docker\models\containers.py in run(self, image, command, stdout, stderr, remove, **kwargs)
    812                                     detach=detach, **kwargs)
    813         except ImageNotFound:
--> 814             self.client.images.pull(image, platform=platform)
    815             container = self.create(image=image, command=command,
    816                                     detach=detach, **kwargs)

~\anaconda3\lib\site-packages\docker\models\images.py in pull(self, repository, tag, all_tags, **kwargs)
    444             del kwargs['stream']
    445 
--> 446         pull_log = self.client.api.pull(
    447             repository, tag=tag, stream=True, all_tags=all_tags, **kwargs
    448         )

~\anaconda3\lib\site-packages\docker\api\image.py in pull(self, repository, tag, stream, auth_config, decode, platform, all_tags)
    428         )
    429 
--> 430         self._raise_for_status(response)
    431 
    432         if stream:

~\anaconda3\lib\site-packages\docker\api\client.py in _raise_for_status(self, response)
    268             response.raise_for_status()
    269         except requests.exceptions.HTTPError as e:
--> 270             raise create_api_error_from_http_exception(e)
    271 
    272     def _result(self, response, json=False, binary=False):

~\anaconda3\lib\site-packages\docker\errors.py in create_api_error_from_http_exception(e)
     29         else:
     30             cls = NotFound
---> 31     raise cls(e, response=response, explanation=explanation)
     32 
     33 

NotFound: 404 Client Error for http+docker://localnpipe/v1.41/images/create?tag=latest&fromImage=spikeinterface%2Fmountainsort4-base: Not Found ("manifest for spikeinterface/mountainsort4-base:latest not found: manifest unknown: manifest unknown")

BenATemple, 2021-08-15:
Update 8/15/21: I tried using command line within Docker to download the Mountainsort4 environment (docker pull spikeinterface/mountainsort4-base:1.0.0).

Now, when I run the command for the sorter in Jupyter notebook, I get a different sort of error ("
AttributeError: module 'os' has no attribute 'getuid'"). After googling this error, it seems that it has something to do with harmonizing linux and Windows systems?

In general, I think the section on Dockers in the spikeinterface.readthedocs.io could be much better fleshed out. As it stands, it has a very spare guide and has been hard to follow for a new user. Thanks for any help or ideas anyone has!

**Command in Jupyter Notebook**: sorting_MS = ss.run_mountainsort4(recording_preprocessed, output_folder='results_Tetrode', docker_image = "spikeinterface/mountainsort4-base")

**Error in Jupyter Notebook**
AttributeError                            Traceback (most recent call last)
<ipython-input-28-10df667f87b7> in <module>
----> 1 sorting_MS = ss.run_mountainsort4(recording_preprocessed, output_folder='results_Tetrode', docker_image = "spikeinterface/mountainsort4-base:1.0.0")

~\anaconda3\lib\site-packages\spikeinterface-0.90.0.dev0-py3.8.egg\spikeinterface\sorters\runsorter.py in run_mountainsort4(*args, **kwargs)
    250 
    251 def run_mountainsort4(*args, **kwargs):
--> 252     return run_sorter('mountainsort4', *args, **kwargs)
    253 run_mountainsort4.__doc__ = _common_run_doc.format('mountainsort4')
    254 

~\anaconda3\lib\site-packages\spikeinterface-0.90.0.dev0-py3.8.egg\spikeinterface\sorters\runsorter.py in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, **sorter_params)
     55             verbose=verbose, raise_error=raise_error, **sorter_params)
     56     else:
---> 57         sorting = run_sorter_docker(sorter_name, recording, docker_image, output_folder=output_folder,
     58                 remove_existing_folder=remove_existing_folder, delete_output_folder=delete_output_folder,
     59                 verbose=verbose, raise_error=raise_error, **sorter_params)

~\anaconda3\lib\site-packages\spikeinterface-0.90.0.dev0-py3.8.egg\spikeinterface\sorters\runsorter.py in run_sorter_docker(sorter_name, recording, docker_image, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, **sorter_params)
    201 
    202     # put file permission to user (because docker is root...)
--> 203     uid = os.getuid()
    204     cmd = f'chown {uid}:{uid} -R /sorting_output_folder/{folder_name}'
    205     commands.append(cmd)

AttributeError: module 'os' has no attribute 'getuid'

alejoe91, 2021-08-17:
Hi Ben! Thanks for trying the docker image and submitting the issue. Indeed this seems to be a big either on the SI side or inside the docker. I'll look into it next week as I'm currently on vacation :) feel free to upgrade the docs and submit a pull request! It would be very much appreciated!

BenATemple, 2021-08-18:
Thanks! I will tinker around a bit more :)

BenATemple, 2021-08-20:
Just a quick update: I've now downloaded the Docker images for Tridesclous, SpykingCircus, Kilosort3, Kilosort2, and Mountainsort 4. And all give me the same error [AttributeError: module 'os' has no attribute 'getuid'], so it's clearly not an issue just with the Mountainsort docker -- likely something in SI? Thanks again for your help!

alejoe91, 2021-08-21:
@BenATemple It seems a windows related problem unfortunately..we'll look into it but so far the docker mechanism has only been tested on Unix systems

BenATemple, 2021-08-22:
Thanks! That's as I suspected. If you guys can get it working in windows that'd be fantastic, and I imagine that it would increase user base. My current machine is windows -- though maybe a Linux simulator (Ubuntu?) is a backup option in the case that the dockers can't be integrated with Windows? From my point of view, the dockers are really really helpful, as they make the across-sorter comparisons more feasible :)

samuelgarcia, 2021-08-23:
I guess that having docker supoort on windows host will be quick challenging.
It is written with linux host in mind only !!!
Stuff like bindings path and chown back to the user are typaclly linux related.

BenATemple, 2021-08-23:
I see -- thanks a lot, Samuel!
What do you recommend as a work-around for someone using SpikeInterface on a Windows computer? Ubuntu?

samuelgarcia, 2021-08-24:
On window you can activate the "linux sub system" it is a terminal that is embed a ubuntu and that have all linux command.
I never play with it so far.
Maybe you can start with that.

But honestly, for spike sorting, I would install a "big" machine somewhere in your lab, with a big GPU. and based on ubuntu. You could do ssh access to run remote computing.

samuelgarcia, 2021-09-17:
I close this.

closed 2021-09-21 by alejoe91