SpikeInterface/spikeinterface #225: Error with NwbRecordingExtractor.save() when reading from remote files (ros3)

luiztauffer, 2021-08-18:
Relevant part of error track:
```python
FileNotFoundError: [Errno 2] Unable to open file (unable to open file: name = '/home/jovyan/https:/dandiarchive.s3.amazonaws.com/blobs/aa6/9ee/aa69ee06-daa9-4f41-af91-4402816d32f2', errno = 2, error message = 'No such file or directory', flags = 0, o_flags = 0)
```
When the correct file path should be: `https://dandiarchive.s3.amazonaws.com/blobs/aa6/9ee/aa69ee06-daa9-4f41-af91-4402816d32f2`
That is, without the prepend `/home/jovyan/`

Full error track:
```python
---------------------------------------------------------------------------
FileNotFoundError                         Traceback (most recent call last)
/tmp/ipykernel_201/2577243958.py in <module>
      1 # Save processed recording locally
----> 2 recording_processed.save(folder='./my_recording')

/opt/conda/envs/ros3/lib/python3.8/site-packages/spikeinterface/core/base.py in save(self, **kwargs)
    517             return self.save_to_memory(**kwargs)
    518         else:
--> 519             return self.save_to_folder(**kwargs)
    520 
    521     def save_to_memory(self, **kwargs):

/opt/conda/envs/ros3/lib/python3.8/site-packages/spikeinterface/core/base.py in save_to_folder(self, name, folder, dump_ext, verbose, **save_kwargs)
    592 
    593         # save data (done the subclass)
--> 594         cached = self._save(folder=folder, verbose=verbose, **save_kwargs)
    595 
    596         self.save_metadata_to_folder(folder)

/opt/conda/envs/ros3/lib/python3.8/site-packages/spikeinterface/core/baserecording.py in _save(self, format, **save_kwargs)
    133 
    134             job_kwargs = {k: save_kwargs[k] for k in self._job_keys if k in save_kwargs}
--> 135             write_binary_recording(self, file_paths=file_paths, dtype=dtype, **job_kwargs)
    136 
    137             from .binaryrecordingextractor import BinaryRecordingExtractor

/opt/conda/envs/ros3/lib/python3.8/site-packages/spikeinterface/core/core_tools.py in write_binary_recording(recording, file_paths, dtype, add_file_extension, verbose, byte_offset, **job_kwargs)
    234     executor = ChunkRecordingExecutor(recording, func, init_func, init_args, verbose=verbose,
    235                                       job_name='write_binary_recording', **job_kwargs)
--> 236     executor.run()
    237 
    238 

/opt/conda/envs/ros3/lib/python3.8/site-packages/spikeinterface/core/job_tools.py in run(self)
    214                 all_chunks = tqdm(all_chunks, ascii=True, desc=self.job_name)
    215 
--> 216             worker_ctx = self.init_func(*self.init_args)
    217             for segment_index, frame_start, frame_stop in all_chunks:
    218                 res = self.func(segment_index, frame_start, frame_stop, worker_ctx)

/opt/conda/envs/ros3/lib/python3.8/site-packages/spikeinterface/core/core_tools.py in _init_binary_worker(recording, rec_memmaps_dict, dtype)
    153     if isinstance(recording, dict):
    154         from spikeinterface.core import load_extractor
--> 155         worker_ctx['recording'] = load_extractor(recording)
    156     else:
    157         worker_ctx['recording'] = recording

/opt/conda/envs/ros3/lib/python3.8/site-packages/spikeinterface/core/base.py in load_extractor(file_or_folder_or_dict, base_folder)
    756     """
    757     if isinstance(file_or_folder_or_dict, dict):
--> 758         return BaseExtractor.from_dict(file_or_folder_or_dict, base_folder=base_folder)
    759     else:
    760         return BaseExtractor.load(file_or_folder_or_dict, base_folder=base_folder)

/opt/conda/envs/ros3/lib/python3.8/site-packages/spikeinterface/core/base.py in from_dict(d, base_folder)
    301             assert base_folder is not None, 'When  relative_paths=True, need to provide base_folder'
    302             d = _make_paths_absolute(d, base_folder)
--> 303         extractor = _load_extractor_from_dict(d)
    304         folder_metadata = d.get('folder_metadata', None)
    305         if folder_metadata is not None:
[... 75 lines cut ...]
    120         return file_obj
    121 

/opt/conda/envs/ros3/lib/python3.8/site-packages/h5py/_hl/files.py in __init__(self, name, mode, driver, libver, userblock_size, swmr, rdcc_nslots, rdcc_nbytes, rdcc_w0, track_order, fs_strategy, fs_persist, fs_threshold, **kwds)
    442             with phil:
    443                 fapl = make_fapl(driver, libver, rdcc_nslots, rdcc_nbytes, rdcc_w0, **kwds)
--> 444                 fid = make_fid(name, mode, userblock_size,
    445                                fapl, fcpl=make_fcpl(track_order=track_order, fs_strategy=fs_strategy,
    446                                fs_persist=fs_persist, fs_threshold=fs_threshold),

/opt/conda/envs/ros3/lib/python3.8/site-packages/h5py/_hl/files.py in make_fid(name, mode, userblock_size, fapl, fcpl, swmr)
    197         if swmr and swmr_support:
    198             flags |= h5f.ACC_SWMR_READ
--> 199         fid = h5f.open(name, flags, fapl=fapl)
    200     elif mode == 'r+':
    201         fid = h5f.open(name, h5f.ACC_RDWR, fapl=fapl)

h5py/_objects.pyx in h5py._objects.with_phil.wrapper()

h5py/_objects.pyx in h5py._objects.with_phil.wrapper()

h5py/h5f.pyx in h5py.h5f.open()

FileNotFoundError: [Errno 2] Unable to open file (unable to open file: name = '/home/jovyan/https:/dandiarchive.s3.amazonaws.com/blobs/aa6/9ee/aa69ee06-daa9-4f41-af91-4402816d32f2', errno = 2, error message = 'No such file or directory', flags = 0, o_flags = 0)
```

alejoe91, 2021-08-18:
Hi @luiztauffer 

This could be a tricky one due to the subprocesses trying to open the file..can you try with 1 job only? We should definitely take a look at this behavior!

luiztauffer, 2021-08-18:
I'll take a look at this for now, go enjoy your rest, @alejoe91 !

luiztauffer, 2021-08-18:
[this commit](https://github.com/catalystneuro/spikeinterface/commit/efb341e5941eeb5e9d97cc927e7c4598e056be5c) fix this issue temporarily, but it's definitely not the best solution.
The best solution will be to make sure the file path is passed without parent folder when it is an s3 path, and that the driver argument gets properly passed forward in all operations

samuelgarcia, 2021-08-23:
Oups.
The http case for file_path was not handle at all in docker.

@alejoe91 : I have in mind something else instead modifying recuirsvely the file path which is a complicated stuff and error prone.
Why not binding the exact entire path for the docker ?
It would much much simpler.

alejoe91, 2022-01-13:
@samuelgarcia is this fixed?

@luiztauffer can you make a PR with the `ros3` argument in th NWB extractors?