SpikeInterface/spikeinterface #226: Error Running SpykingCircus

BenATemple, 2021-08-22:
I get the following error when running Spyking Circus:
AttributeError: module 'circus' has no attribute '__version__'

Does anyone else experience the same error?
(Here's what I enter to run:  ss.run_spykingcircus(recording_preprocessed))

samuelgarcia, 2021-08-23:
This seems related to spyking-circus itself.
Whet happend you you do
```
import circus
print(circus.__version__)
```
On my side it is working.
Did you install spykingcircus from source or from pypi ?

BenATemple, 2021-08-23:
Thanks a lot, Samuel! When I run: ss.installed_sorters, I see 'spykingcircus.'
However, when I run print(circus._version_), I receive: AttributeError: module 'circus' has no attribute '__version__'

I installed SpyKing Circus with Conda. Since it appears when I query installedsorters, I think that it is there. Just not sure why I'm getting this version error.

In general, I think it would help if the method for attaching sorters to SpikeInterface was better fleshed out in the documentation. I've looked on spikeinterface.readthedocs.io and have also gone through the Sep.2020 NWB tutorial but am still a little unclear on how SI is recognizing sorters/how to ensure that this takes place. Thanks again for your help :)

samuelgarcia, 2021-08-24:
Hi,
@yger made a sofisticated stuff for circus version that depend on git. See this:
https://github.com/spyking-circus/spyking-circus/blob/master/circus/_version.py
My guess is that it do not work on your platform.
Pierre can you help Ben ?

Note that Alessio have started a new tutorial with new API but not finished yet see this:
https://github.com/SpikeInterface/spiketutorials/pull/22

To check if sorter is installed depend on the sorter itself, for python based sorter it is simple we just try to `import` See https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/sorters/spyking_circus/spyking_circus.py#L65

yger, 2021-08-24:
I can help, but running print(circus.__version__) and not print(circus.version) should work. Can you confirm that?

closed 2022-01-07 by alejoe91