SpikeInterface/spikeinterface #229: Issues with Loading Neuralynx Files

adityas1997, 2021-08-23:
Hi,

I'm trying to load neuralynx files and receiving the error message copied below:

IndexError                                Traceback (most recent call last)
<ipython-input-1-6a19b36aa9af> in <module>
     48 
     49 rawNeuralynx = '/home/aditya/Raw_Data/Spike_Sorting/Neuralynx_Test/'
---> 50 sortedSpikeTrainArray = buildSpikeTrainArrayFromRecording(rawNeuralynx,'/home/aditya/tmp2/Neura_Test/')
     51 blockSpikeArray = buildArrayOutOfSpikeTrain(sortedSpikeTrainArray)
     52 print(np.shape(blockSpikeArray))

<ipython-input-1-6a19b36aa9af> in buildSpikeTrainArrayFromRecording(recordingDir, toSaveToDir)
     15         recording, sorting_true = se.toy_example(duration=10, seed=0, num_segments=1)
     16     else:
---> 17         recording = se.read_neuralynx(recordingDir,stream_id=None)
     18     recording_cached = recording.save(name=toSaveToDir,overwrite=True)
     19     sortingHDS = ss.run_sorter(sorter_name='herdingspikes', recording=recording_cached,

/usr/bin/anaconda3/lib/python3.7/site-packages/spikeinterface/extractors/neoextractors/neuralynx.py in read_neuralynx(*args, **kwargs)
     27 
     28 def read_neuralynx(*args, **kwargs):
---> 29     recording = NeuralynxRecordingExtractor(*args, **kwargs)
     30     return recording
     31 

/usr/bin/anaconda3/lib/python3.7/site-packages/spikeinterface/extractors/neoextractors/neuralynx.py in __init__(self, folder_path, stream_id)
     21     def __init__(self, folder_path, stream_id=None):
     22         neo_kwargs = {'dirname': folder_path}
---> 23         NeoBaseRecordingExtractor.__init__(self, stream_id=stream_id, **neo_kwargs)
     24 
     25         self._kwargs = dict(folder_path=folder_path, stream_id=stream_id)

/usr/bin/anaconda3/lib/python3.7/site-packages/spikeinterface/extractors/neoextractors/neobaseextractor.py in __init__(self, stream_id, **neo_kwargs)
     36                 raise ValueError(f'This reader have several streams({stream_ids}), specify it with stream_id=')
     37             else:
---> 38                 stream_id = stream_ids[0]
     39         else:
     40             assert stream_id in stream_ids, f'stream_id {stream_id} is no in {stream_ids}'

IndexError: index 0 is out of bounds for axis 0 with size 0

Any suggestions would be appreciated.

samuelgarcia, 2021-08-24:
Hi I think the problem is in neo itself.
You try:
```
import neo
reader = neo.NeuralynxIO( '/home/aditya/Raw_Data/Spike_Sorting/Neuralynx_Test/')
print(reader)
```

Could you share your file also ?

adityas1997, 2021-08-24:
[Neura_Test.zip](https://github.com/SpikeInterface/spikeinterface/files/7039512/Neura_Test.zip)

samuelgarcia, 2021-08-24:
The file you send me is OK with neo but it is only the"nse" file that contain only spikes.
Traces are in "nsc" files.
In spikeinterface recording object are traces (signals), this is why it bugs.

samuelgarcia, 2021-09-17:
Does the anwser helps ? Can I close ?

alejoe91, 2021-09-17:
> Does the anwser helps ? Can I close ?

I think you can close this @samuelgarcia