SpikeInterface/spikeinterface #233: 'mountainsort4' has no attribute 'mountainsort4_curation'

Feikai-Lin, 2021-08-25:
Hi, 

I was using mountainsort4 1.0.0 via spikeinterface 0.90.0 to sort my data with jupyter notebook. I have also installed ml-ms4alg. The sorter worked properly but it gave an error when I changed the default parameter: curation, to True, and the value of noise_overlap_threshold. Here is what I got in the log file: 

"error_trace": "Traceback (most recent call last):\n  File \"/home/lenovo/anaconda3/envs/ephy/lib/python3.8/site-packages/spikeinterface/sorters/basesorter.py\", line 198, in run_from_folder\n    SorterClass._run_from_folder(output_folder, sorter_params, verbose)\n  File \"/home/lenovo/anaconda3/envs/ephy/lib/python3.8/site-packages/spikeinterface/sorters/mountainsort4/mountainsort4.py\", line 133, in _run_from_folder\n    old_api_sorting = mountainsort4.mountainsort4_curation(\nAttributeError: module 'mountainsort4' has no attribute 'mountainsort4_curation'\n",

In spikeinterface/sorters/mountainsort4/mountainsort4.py line 133 it calls 'mountainsort4_curation(), but mountainsort4 seems not to have this attribute. I did find mountainsort4_curation() as a function in package ml_ms4alg (ml_ms4alg/ml_ms4alg/mountainsort4_curation.py). Is it a bug in the code? Can someone help me with it?

Really appreciate it!

alejoe91, 2021-08-25:
Hi @Feikai-Lin 

Thanks for the issue, it seems that the new `mountainsort4` doesn't have the curation feature. @magland can you confirm? In that case, I'll remove that option from the parameters and the code.

samuelgarcia, 2021-08-25:
You also to uinstall   `ml-ms4alg` (https://pypi.org/project/ml-ms4alg/) package and install `mountainsort4` package instead (https://pypi.org/project/mountainsort4/)

alejoe91, 2021-08-25:
> You also to uinstall `ml-ms4alg` (https://pypi.org/project/ml-ms4alg/) package and install `mountainsort4` package instead (https://pypi.org/project/mountainsort4/)

@samuelgarcia the problem is in the new `mountainsort4`! [This curaiton function](https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/sorters/mountainsort4/mountainsort4.py#L133) doesn't exist in `mountainsort4`

samuelgarcia, 2021-08-25:
Sorry I see.
We should remove theses line and this option so.
Lets wait @magland confirmation.

closed 2021-08-25 by alejoe91