SpikeInterface/spikeinterface #252: Bug with PhyKilosortExtractors

yger, 2021-09-07:
The object assumes that .tsv files are there with data on the clusters. This is not the case, at least for KS1.  

/media/cure/Secondary/pierre/softwares/spikeinterface/spikeinterface/extractors/phykilosortextractors.py in __init__(self, folder_path, exclude_cluster_groups, keep_good_only)
     72                     cluster_info = pd.merge(cluster_info, new_property, on='cluster_id')
     73 
---> 74             cluster_info["id"] = cluster_info["cluster_id"]
     75             del cluster_info["cluster_id"]
     76 

TypeError: 'NoneType' object is not subscriptable

alejoe91, 2021-09-07:
Waiting for your PR :)

alejoe91, 2021-09-08:
https://github.com/SpikeInterface/spikeinterface/pull/250

closed 2021-09-08 by alejoe91