SpikeInterface/spikeinterface #253: Chose template mode when exporting to phy

Dradeliomecus, 2021-09-07:
Hi,

When using the `export_to_phy` function, the template mode will automatically be `'median'`, and I don't believe there is an argument to change it.

Could it be possible to either:
- Have an argument in the `export_to_phy` function to specify the template mode
- Set the default template mode in `WaveformExtractor.set_params()` (which would be really nice I think)

The problem for me is that the median is very computationally expensive for large clusters (> 300,000 spikes) and takes a lot of time, whereas I am ok with the mean (which can save me tens of minutes).

Thank you!

alejoe91, 2021-09-07:
@Dradeliomecus thanks for the issue! I think the first option is better (having an extra param in phy) because the `WaveformExtractor._params` actually only control how to extract the waveforms, so they are not really related to the template. 

Would you have time to give it a tray and make a PR for it?

Cheers
Alessio

Dradeliomecus, 2021-09-07:
Thanks for the quick answer!

Yes I can make a PR for it, but not right now.
When I'll have some free time, I'll make the PR and link it to this issue.

Btw great job with the refactoring, it looks really great!

alejoe91, 2021-09-07:
Thank you @Dradeliomecus! We are glad you're liking it!

Most of the work was done by @samuelgarcia :)

samuelgarcia, 2021-09-07:
@Dradeliomecus : I guess you owe a beer. I sometime go to your lab. I propose a the pub Gay Lussac :)

Dradeliomecus, 2021-09-07:
Sure, whenever you want! That would be nice.

By the way, I've been searching but I was unable to find the `CurationSortingExtractor`.
Is there an equivalent in the new version to merge clusters, change their spike times, delete clusters etc... ?

alejoe91, 2021-09-07:
@Dradeliomecus not really, but that is something that could be useful to port. @samuelgarcia what do you think?

samuelgarcia, 2021-09-08:
CurationSortingExtractor can be ported but it will be a bit tricky to make an true extractor because in the new API the unit_ids cannot be anymore changed on the fly. Maybe with some hack at Base level it coule OK.

Anyway we can have a similar obect that behave like an sortingextarctor that could be usefull to build a GUI on top of it.

Dradeliomecus, 2021-09-08:
@alejoe91 @samuelgarcia Thank you for the quick and helpful answers!

I think I'll make a seperate issue for it to keep things clear.