SpikeInterface/spikeinterface #254: Importing CurationSortingExtractor to new version

Dradeliomecus, 2021-09-08:
Hi,

Following the discussion in #253:

It would be really nice to have something resembling the `CurationSortingExtractor`, allowing us to delete units, merge some, change their spike timings etc...

Technically I think the only base methods that are required are removing a unit, and changing the spike train.
Then merging and other methods can probably be made from those 2.

This is something crucial for what I'm doing.
I really want to use the new version, but this is something important for what I'm doing.

Thanks for all the good work!

samuelgarcia, 2021-09-08:
I let check the new design in the code. We now handle multi segment.
By inheritance I think it is possible to do it.
If you want I propose a short before you start the implementation.

Dradeliomecus, 2021-09-08:
> I let check the new design in the code. We now handle multi segment.
> By inheritance I think it is possible to do it.
> If you want I propose a short before you start the implementation.

I'm sorry, but I don't quite understand what you just said.

alejoe91, 2021-09-08:
> > I let check the new design in the code. We now handle multi segment.
> > By inheritance I think it is possible to do it.
> > If you want I propose a short before you start the implementation.
> 
> I'm sorry, but I don't quite understand what you just said.

@Dradeliomecus sometimes @samuelgarcia needs a translation :P 

"I let check the new design in the code. We now handle multi segment."
-->
"I'll let you check the new code design, that now handles multiple segments in the same recording/sorting object."

"By inheritance I think it is possible to do it."
-->
"Probably, it could be possible to implement the CurationSortingExtractor by inheritance"

"If you want I propose a short before you start the implementation."
-->
"If you want, I propose a short MEETING before you start the implementation"

Hope it's clearer :)

Dradeliomecus, 2021-09-08:
Ah yes, thank you.

Yes I agree, a meeting would be a great idea!

samuelgarcia, 2021-09-08:
:)

Dradeliomecus, 2021-09-20:
@samuelgarcia When do you propose we do the short meeting?

We can either do it online or when you come at IBENS (I owe you a beer after all :) )

samuelgarcia, 2021-09-21:
Hi,
I am in Lyon, I go to IBENS once a year in average.
could you send me a direct email to setup a call online ?

Dradeliomecus, 2021-09-21:
You can contact me at: aurelien.wyngaard@ens.fr

ferchaure, 2022-04-12:
any update about this? I need a way to merge units and maybe I can think how to implement it.

samuelgarcia, 2022-04-13:
Hi Fernando,
no news on this.
Alessio was super motivated to do it but he is now on the fight "file compression".
Maybe he will do it.
This can be a project for the hackthon in June :)

alejoe91, 2022-04-13:
Hola @ferchaure 

At least I have given some implementation thoughts. Maybe we could chat about it? Send me an email :)

alejoe91, 2022-08-01:
Done by @ferchaure

closed 2022-08-01 by alejoe91