SpikeInterface/spikeinterface #260: read_tridesclous

farazmoradi, 2021-09-11:
Hi, 

I have updated my spike interface and I'm getting errors while reading sorting results. this did not happen in the previous version.

sorting = se.read_tridesclous('/Volumes/Project/tdc_d1_b1_TG1_4')

TypeError: 'NoneType' object is not subscriptable
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-11-0cf1d6571019> in <module>
----> 1 sorting = se.read_tridesclous('/Volumes/Project/tdc_d1_b1_TG1_4')

~/opt/anaconda3/lib/python3.8/site-packages/spikeinterface-0.90.2.dev0-py3.8.egg/spikeinterface/extractors/tridesclousextractors.py in read_tridesclous(*args, **kwargs)
     61 
     62 def read_tridesclous(*args, **kwargs):
---> 63     sorting = TridesclousSortingExtractor(*args, **kwargs)
     64     return sorting
     65 

~/opt/anaconda3/lib/python3.8/site-packages/spikeinterface-0.90.2.dev0-py3.8.egg/spikeinterface/extractors/tridesclousextractors.py in __init__(self, folder_path, chan_grp)
     29         catalogue = dataio.load_catalogue(name='initial', chan_grp=chan_grp)
     30 
---> 31         labels = catalogue['clusters']['cluster_label']
     32         labels = labels[labels >= 0]
     33         unit_ids = list(labels)

TypeError: 'NoneType' object is not subscriptable

samuelgarcia, 2021-09-13:
Hi Faraz,
This is strange read_tridesclous have been changed for a while. When did you run tridesclous one that folder ?

Could you try to re run a separate tridesclous and check that do read the output to verify that the bug is due to spikeinterface ?

farazmoradi, 2021-09-13:
I did this and it worked. I think the problem is with the spike interface

farazmoradi, 2021-09-14:
If this is not working, do you have any idea about how to load " .mat " outputs of tridesclous in spikeinteraface?

closed 2021-09-17 by samuelgarcia