SpikeInterface/spikeinterface #261: Failure of export_to_phy from existing mountainsort cluster data

SynapticSage, 2021-09-11:
I have a use case where I take already data already sorted and curated with MountainSort, and funnel through SpikeInterface  API to generate phy files...

```python
import json, os, spikeinterface
import spikeinterface.core.waveform_extractor as wave
import spikeinterface.exporters.to_phy as phy
import spikeinterface.extractors.mdaextractors as mda

# Point to a prv recording and the params file
local_path = '/Volumes/GenuDrive/RY16_direct/MountainSort/RY16_36.mountain/RY16_36.nt1.mountain'
prv_file = os.path.join(local_path, 'raw.mda.prv')
params_file = os.path.join(local_path, 'params.json')

# Derive the properties in those json dicts
with open(prv_file, 'r') as F:
    raw_dict = json.load(F)
with open(params_file, 'r') as F:
    params_dict = json.load(F)

# Create the mda object
mda_reecording = mda.read_mda_recording(local_path,
                                   raw_fname=raw_dict['original_path'])
mda_reecording.annotate(is_filtered=True) # we've already filtered it, so mark it thus

# Derive the spikes file
firings_file = os.path.join(local_path, 'firings_raw.mda')
sorted_spikes = mda.read_mda_sorting(firings_file, sampling_frequency=params_dict['samplerate'])

# Create waveforms per spike
waveform_file = os.path.join(local_path, 'waveforms')
waveforms = wave.extract_waveforms(mda_reecording, sorted_spikes, waveform_file)

# Export to phy
phyplace = os.path.join(local_path, "phy")
phy.export_to_phy(waveforms, phyplace)
```

Which is great: I like phy a bit more than qt-mountainview curation. but this can cause an error in the export_phy step. namely, the windows of the waveforms, the end_frame points can overshoot the mda N1 x N2 size bounds. I think it requires checks starting in the top call to ensure bounds are never exceeded. 

In other worrds, it works if I add this line in job_tools.py

```python
def devide_recording_into_chunks(recording, chunk_size):
 51     all_chunks = []
 52     for segment_index in range(recording.get_num_segments()):
 53         num_frames = recording.get_num_samples(segment_index)
 54         chunks = divide_segment_into_chunks(num_frames, chunk_size)
 55         #all_chunks.extend([(segment_index, frame_start, frame_stop) for frame_start, frame_stop in chunks])
 56         sample_max = recording.get_num_samples()
 57         all_chunks.extend([(segment_index, frame_start, np.min((frame_stop, sample_max)) for frame_start, frame_stop in chunks])
 58     return all_chunks
```

alejoe91, 2021-09-12:
Hi @SynapticSage 

Thanks for submitting the issue!

Can you also parlate the error that you get?? In principle, even using an `end_frame` larger than num samples should not trigger an error, so maybe the MDA recording extractor needs a modification

SynapticSage, 2021-09-12:
Yep. Attaching:

```python
[ins] In [28]: phyfiles = phy.export_to_phy(waveform, local_path + os.path.sep + "phy")
write_binary_recording with n_jobs 1  chunk_size None
Setting 'return_scaled' to False
-------------------------------------------------------------------------
ValueError                              Traceback (most recent call last)
<ipython-input-28-66a465ff0240> in <module>
----> 1 phyfiles = sexpphy.export_to_phy(waveform, local_path + os.path.sep + "phy")

~/anaconda3/envs/phy2/lib/python3.7/site-packages/spikeinterface-0.90.2.dev0-py3.7.egg/spikeinterface/exporters/to_phy.py in export_to_phy(waveform_extractor, output_folder, compute_pc_features, compute_amplitudes, max_channels_per_template, copy_binary, remove_if_exists, peak_sign, template_mode, dtype, verbose, **job_kwargs)
    166         pc.run_for_all_spikes(output_folder / 'pc_features.npy',
    167                               max_channels_per_template=max_channels_per_template, peak_sign=peak_sign,
--> 168                               **job_kwargs)
    169
    170         max_channels_per_template = min(max_channels_per_template, len(channel_ids))

~/anaconda3/envs/phy2/lib/python3.7/site-packages/spikeinterface-0.90.2.dev0-py3.7.egg/spikeinterface/toolkit/postprocessing/principal_component.py in run_for_all_spikes(self, file_path, max_channels_per_template, peak_sign, **job_kwargs)
    227         init_args = init_args + (all_pcs_args, spike_times, spike_labels, we.nbefore, we.nafter, unit_channels, all_pca)
    228         processor = ChunkRecordingExecutor(recording, func, init_func, init_args, job_name='extract PCs', **job_kwargs)
--> 229         processor.run()
    230
    231     def _fit_by_channel_local(self):

~/anaconda3/envs/phy2/lib/python3.7/site-packages/spikeinterface-0.90.2.dev0-py3.7.egg/spikeinterface/core/job_tools.py in run(self)
    220             worker_ctx = self.init_func(*self.init_args)
    221             for segment_index, frame_start, frame_stop in all_chunks:
--> 222                 res = self.func(segment_index, frame_start, frame_stop, worker_ctx)
    223                 if self.handle_returns:
    224                     returns.append(res)

~/anaconda3/envs/phy2/lib/python3.7/site-packages/spikeinterface-0.90.2.dev0-py3.7.egg/spikeinterface/toolkit/postprocessing/principal_component.py in _all_pc_extractor_chunk(segment_index, start_frame, end_frame, worker_ctx)
    354     start = int(spike_times[i0] - nbefore)
    355     end = int(spike_times[i1 - 1] + nafter)
--> 356     traces = recording.get_traces(start_frame=start, end_frame=end, segment_index=segment_index)
    357
    358     for i in range(i0, i1):

~/anaconda3/envs/phy2/lib/python3.7/site-packages/spikeinterface-0.90.2.dev0-py3.7.egg/spikeinterface/core/baserecording.py in get_traces(self, segment_index, start_frame, end_frame, channel_ids, order, return_scaled)
     98         channel_indices = self.ids_to_indices(channel_ids, prefer_slice=True)
     99         rs = self._recording_segments[segment_index]
--> 100         traces = rs.get_traces(start_frame=start_frame, end_frame=end_frame, channel_indices=channel_indices)
    101         if order is not None:
    102             assert order in ["C", "F"]

~/anaconda3/envs/phy2/lib/python3.7/site-packages/spikeinterface-0.90.2.dev0-py3.7.egg/spikeinterface/extractors/mdaextractors.py in get_traces(self, start_frame, end_frame, channel_indices)
    129             end_frame = self.get_num_samples()
    130         recordings = self._diskreadmda.readChunk(i1=0, i2=start_frame, N1=self._diskreadmda.N1(),
--> 131                                                  N2=end_frame - start_frame)
    132         recordings = recordings[channel_indices, :].T
    133         return recordings

~/anaconda3/envs/phy2/lib/python3.7/site-packages/spikeinterface-0.90.2.dev0-py3.7.egg/spikeinterface/extractors/mdaextractors.py in readChunk(self, i1, i2, i3, N1, N2, N3)
    342                 A = np.load(self._path, mmap_mode='r')
    343                 return A[:, i2:i2 + N2]
--> 344             return np.reshape(X, (N1, N2), order='F')
    345         else:
    346             if N1 != self.N1():

<__array_function__ internals> in reshape(*args, **kwargs)

~/anaconda3/envs/phy2/lib/python3.7/site-packages/numpy/core/fromnumeric.py in reshape(a, newshape, order)
    297            [5, 6]])
    298     """
--> 299     return _wrapfunc(a, 'reshape', newshape, order=order)
    300
    301

~/anaconda3/envs/phy2/lib/python3.7/site-packages/numpy/core/fromnumeric.py in _wrapfunc(obj, method, *args, **kwds)
[... 60 lines cut ...]
    343                 return A[:, i2:i2 + N2]
--> 344             return np.reshape(X, (N1, N2), order='F')
    345             #return np.reshape(X, (N1-i1, N2-i2), order='F')
    346         else:

<__array_function__ internals> in reshape(*args, **kwargs)

~/anaconda3/envs/phy2/lib/python3.7/site-packages/numpy/core/fromnumeric.py in reshape(a, newshape, order)
    297            [5, 6]])
    298     """
--> 299     return _wrapfunc(a, 'reshape', newshape, order=order)
    300
    301

~/anaconda3/envs/phy2/lib/python3.7/site-packages/numpy/core/fromnumeric.py in _wrapfunc(obj, method, *args, **kwds)
     56
     57     try:
---> 58         return bound(*args, **kwds)
     59     except TypeError:
     60         # A TypeError occurs if the object does have such a method in its

ValueError: cannot reshape array of size 2053833200 into shape (4,513585503)
```

I applied the same bandage for that too on my end at the extractor level. But before doing that earlier today, looked around the mda extractor...but my eye didn't catch anything obviously false --- although couldn't tell if the header offset was correct.

alejoe91, 2021-09-12:
I think this could be fixed by setting `end_frame=num_samples` if `end_frame>num_samples` in the `get_traces` function of the `MdaRecordingSegment` :)

samuelgarcia, 2021-09-13:
@alejoe91 : I don't agree!!!
For me if `end_frame>num_samples` we need to raise an error.

Here the error is in waveform extractor that try get sample outside the zone and we need to handle the corner case there.

alejoe91, 2021-09-13:
ok, then can you fix the `ensure_chunk_size`?

samuelgarcia, 2021-09-13:
I will have a look for the bug is really in waveform extarctor.

If a spike is near border we shoudl return a 0 waveforms (or partialy filled woth zeros)

@SynapticSage : could you generate for me a fake recording and sorting that trigger the same bug ?

alejoe91, 2021-09-13:
@samuelgarcia he already shared the MDA files [here](https://drive.google.com/drive/u/1/folders/17S8T67GIdxemwXfzBpEfusESnfrXs7Kh)

samuelgarcia, 2021-09-13:
OK sorry.
io will have a look.

samuelgarcia, 2021-09-13:
Oups the link is not valid anymore. @SynapticSage : can you share again files ?

SynapticSage, 2021-09-13:
@samuelgarcia It's back [up](https://drive.google.com/drive/u/0/folders/1yR26-xmt-tCw81QeNqkUJ9IU-hCvTBeV) . raw.mda or filt.mda should work as test cases.

samuelgarcia, 2021-09-14:
I try to test but there is no 'geom.csv' file your drive and the MDA reader need it apparently.

SynapticSage, 2021-09-14:
Added `geom.csv` to the share folder.

samuelgarcia, 2021-09-15:
THanks I will chekc this tomorow.

samuelgarcia, 2021-09-16:
Hi.
I am able to reproduce the bug sometimes.

It is at 2 places in fact for the same reason:
  * WaveformExtractor
  * WaveformPrincipalComponent

In your data the trace sample size is `513896382` and the very last spike is  `513896349` so with ms_after=4ms it is 120 sample and so the waveform cannot be extracted.

In WaveformExtractor, theses spike are already exlcude with we "sample" a subset of spikes in select_random_spikes_uniformly().
See here https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/core/waveform_extractor.py#L428

But in case you want ot extract all the spikes (aka `max_spikes_per_unit=None` then it fails)

And the same append in WaveformPrincipalComponent because of the same reason.
I will try a fix for that.

samuelgarcia, 2021-09-16:
I started something here #268

samuelgarcia, 2021-09-16:
This is strange.
In fact this line was already protecting form max_spikes_per_unitNone`  https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/core/waveform_extractor.py#L428

So my patch is totally useless. I really don't anderstand why you have the bug in waveform extractor.

In my case I only the bug in WaveformPrincipalComponent not in WaveformExtractor

samuelgarcia, 2021-09-16:
Hi @SynapticSage 

you have an error in your script : the signal is not filtered but you do `mda_reecording.annotate(is_filtered=True)`
It is a mistake because all waveforms and PCA are totally wrong in that case you need to 
`mda_reecording_f = si.bandpass_filter(mda_reecording,  freq_min=300., freq_max=6000.)`
and use this recording for `extract_waveforms(mda_reecording_f, ...)`

Also note that you speed up a lot if you have python>=3.8 with `n_jobs and chunk_memory/total_memory`
```python
waveforms = wave.extract_waveforms(mda_reecording_f, sorted_spikes, waveform_file, 
  #~ max_spikes_per_unit=None,
  max_spikes_per_unit=500,
  ms_before=3., ms_after=4.,
  return_scaled=False,
  n_jobs=10, total_memory='50M',
  progress_bar=True,
)
```

See https://spikeinterface.readthedocs.io/en/latest/api.html#spikeinterface.core.ChunkRecordingExecutor

And same for phy:

```python
phy.export_to_phy(waveforms, phyplace,
  n_jobs=10, total_memory='50M',
  progress_bar=True)
```

samuelgarcia, 2021-09-16:
Please the #268 PR that shoudl fix teh bug.

SynapticSage, 2021-09-16:
Oops. That was my bad. I accidentally copy-pasted parts from two completely different script files when I made the issue. 

One of the scripts use raw `raw.mda` sans `is_filtered` and one uses `filt.mda` (instead of `raw.mda`) from our pipeline.

We still had the overshoot bug for either case. But I really appreciate the support on this issue!

samuelgarcia, 2021-09-16:
Did you try my working branch ?

```
pip uninstall spikeinterface
git clone https://github.com/samuelgarcia/spikeinterface.git
cd spikeinterface
git checkout some_fixes
python setup.py install
```

On my side there is no bug anymore.

SynapticSage, 2021-09-17:
Just tried it. Sorry for the delay. Running experiments earlier.

Totally works 
👍 Thanks :)

closed 2021-09-17 by SynapticSage