SpikeInterface/spikeinterface #265: Trouble in loading simulated binary data

Jyh-Jang, 2021-09-15:
Dear,

I am wondering whether the following code can make a proper binary file for BinDatRecordingExtractor.
LFP is numpy.ndarray (384, 128001).

newFile = open("/test.bin", "wb")
newFileByteArray = np.array(LFP, dtype= np.int16)
newFile.write(newFileByteArray)

Thank you,
JJ

samuelgarcia, 2021-09-15:
BinDatRecordingExtractor is the old API for spikeextractor in this case the shape is (channel, sample)
Now the spikeinterface  integrate the extractor module and the shape is (sample, channel).

I don't which spikeinterface you have if >=0.90. then it is the new API. if version <=0.13  then it is "old API"

See this
https://spikeinterface.readthedocs.io/en/latest/releases/0.90.0.html

I would suggest to go for new API.

In any case, in spikeinterface write and read are with objects : sorting and recording. You can't use direct numpy array.
You can convert numpy array in theses object with specific classes : NumpyRecording / NumpySorting.

I let you carrfully read the documentation : 
https://spikeinterface.readthedocs.io/en/latest/modules_doc.html
and particularly theses examples:
https://spikeinterface.readthedocs.io/en/latest/modules/index.html

Jyh-Jang, 2021-09-15:
Thanks, samuelgarcia.

We have figured it out and it works with the following commands.

newFile = open("/test.bin", "wb")
newFileByteArray = np.array(LFP, dtype= np.int16)
newFile.write(newFileByteArray.tobytes('F'))

Best,
JJ

closed 2021-10-14 by alejoe91