SpikeInterface/spikeinterface #266: Problem reading legacy OpenEphys format file

ToGauvrit, 2021-09-15:
Hi,
I think the problem come from here:

```
def read_openephys(folder_path, **kwargs):
    """
    Read 'legacy' or 'binary' Open Ephys formats

    Parameters
    ----------
    folder_path: str or Path
        Path to openephys folder

    Returns
    -------
    recording: OpenEphysLegacyRecordingExtractor or OpenEphysBinaryExtractor
    """
    # auto guess format
    files = [str(f) for f in Path(folder_path).iterdir()]
    if np.any([f.startswith('Continuous') for f in files]):
        #  format = 'legacy'
        recording = OpenEphysLegacyRecordingExtractor(folder_path, **kwargs)
    else:
        # format = 'binary'
        recording = OpenEphysBinaryRecordingExtractor(folder_path, **kwargs)
    return recording

```
The var files is a list of path and not just name of files. So the startswith cannot find any file beginning by 'Continuous' and it's always new binary format that is triggered. 
I find a way to solve this like this:
`if np.any([f.startswith('Continuous',len(folder_path)+1) for f in files]):`

Thank you,

Théo

samuelgarcia, 2021-09-15:
I think the bug `startswith()` should be `endswith()` no ?
could you try it ?

ToGauvrit, 2021-09-15:
It's working too with endswith and 'continuous' with a 'c' in a lower case. I couldn't test with the new binary format but I think there is no files ending with continuous, no?
`if np.any([f.endswith('continuous') for f in files]):`

samuelgarcia, 2021-09-15:
Yes the new binary don't have this "*.continuous" files.
Could try a PR ?

alejoe91, 2021-09-21:
https://github.com/SpikeInterface/spikeinterface/pull/269

closed 2021-09-21 by alejoe91