SpikeInterface/spikeinterface #267: quality metric: noise overlap

khl02007, 2021-09-15:
@alejoe91 submitting an issue here first per our discussion. the noise overlap quality metric is still not implemented in spikeinterface>0.90. I can submit a PR addressing this. 

One difference between this metric and other PC based metric is that it requires generating a noise cluster by random sampling of the recording, so needs more than just the PC projection of the waveforms. presumably the recording can be accessed from the waveform extractor

samuelgarcia, 2021-09-15:
hi @khl02007.
Yes you can acces it with
`waveform_extractor.recording`

Please have a look to thais function
https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/toolkit/utils.py#L5

This can be usefull to collect snipet of noise.
We an have modified version of this that avoid taking chunk with spikes (which is not always possible for high rates and hich channel counts)

khl02007, 2021-09-15:
@samuelgarcia cool, seems very useful

khl02007, 2021-10-02:
@samuelgarcia Is there an easy way to recover the principal components from a `WaveformPrincipalComponent` object? Here by principal components I mean the eigenvectors onto which the individual waveforms were projected. I checked `get_components` but it seems that returns the projection onto the eigenvectors, not the eigenvectors themselves. I want this because I'd like to project the noise snippets onto the same principal components as the spikes so that they are in the same feature space for doing the noise overlap computation.

alejoe91, 2021-10-03:
@khl02007 you are very right! I think we should call the current `get_components` either `get_scores` or `get_projections`. What do you recommend? We are going to add a method (maybe keeping the component notation) to return the actual vectors. We also need it for the new gui to compute scores on the fly. I'll ping you when I have a PR ready

khl02007, 2021-10-03:
@alejoe91 Thanks, I personally would prefer `get_projections` as I'm more familiar with that terminology. And yes, it would be good to reserve components for the eigenvectors.

samuelgarcia, 2021-10-04:
@khl02007 : at the moment the projections are not saved persistently on disk.
We will do it quite soon.
If you look at the code there is somewhere a `all_pca` that is a list of sklearn PCA (or any projector) object.The idea is to save this object in the folder.
Note that depending option `get_projections()` should given the matrice of projection but also the "offset" when you substract the mean before the PCA.

khl02007, 2021-10-12:
@alejoe91 @samuelgarcia Hi guys, just wanted to check in and ask when `get_components` might be ready  - would like to submit a PR for the noise overlap metric using it soon.

samuelgarcia, 2021-10-12:
Hi @khl02007. From my side I am still a bit busy and have time to implement this immediatly.
I am sorry. @alejoe91 : do you have more time ?

alejoe91, 2021-10-12:
@samuelgarcia I can try to draft it as we discussed!

closed 2021-11-06 by khl02007