SpikeInterface/spikeinterface #283: `ValueError: numpy.ndarray size changed, may indicate binary incompatibility` when running `Getting Started Tutorial`

marcoistasy, 2021-10-04:
I am running into this errors message when trying to run `print('Installed sorters', ss.installed_sorters())`

```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
/tmp/ipykernel_20952/2839960169.py in <module>
      1 print('Available sorters', ss.available_sorters())
----> 2 print('Installed sorters', ss.installed_sorters())

~/anaconda/envs/visification/lib/python3.7/site-packages/spikeinterface/sorters/sorterlist.py in installed_sorters()
     48     Lists installed sorters.
     49     '''
---> 50     l = sorted([s.sorter_name for s in sorter_full_list if s.is_installed()])
     51     return l
     52 

~/anaconda/envs/visification/lib/python3.7/site-packages/spikeinterface/sorters/sorterlist.py in <listcomp>(.0)
     48     Lists installed sorters.
     49     '''
---> 50     l = sorted([s.sorter_name for s in sorter_full_list if s.is_installed()])
     51     return l
     52 

~/anaconda/envs/visification/lib/python3.7/site-packages/spikeinterface/sorters/mountainsort4/mountainsort4.py in is_installed(cls)
     68     def is_installed(cls):
     69         try:
---> 70             import mountainsort4
     71             HAVE_MS4 = True
     72         except ImportError:

~/anaconda/envs/visification/lib/python3.7/site-packages/mountainsort4/__init__.py in <module>
----> 1 from .mountainsort4 import mountainsort4

~/anaconda/envs/visification/lib/python3.7/site-packages/mountainsort4/mountainsort4.py in <module>
      7 import math
      8 import multiprocessing
----> 9 import spikeextractors as se
     10 
     11 

~/anaconda/envs/visification/lib/python3.7/site-packages/spikeextractors/__init__.py in <module>
      8 from .multisortingextractor import concatenate_sortings, MultiSortingExtractor
      9 
---> 10 from .extractorlist import *
     11 
     12 from . import example_datasets

~/anaconda/envs/visification/lib/python3.7/site-packages/spikeextractors/extractorlist.py in <module>
     19     SpykingCircusRecordingExtractor
     20 from .extractors.spikeglxrecordingextractor.spikeglxrecordingextractor import SpikeGLXRecordingExtractor
---> 21 from .extractors.tridescloussortingextractor.tridescloussortingextractor import TridesclousSortingExtractor
     22 from .extractors.npzsortingextractor.npzsortingextractor import NpzSortingExtractor
     23 from .extractors.mcsh5recordingextractor.mcsh5recordingextractor import MCSH5RecordingExtractor

~/anaconda/envs/visification/lib/python3.7/site-packages/spikeextractors/extractors/tridescloussortingextractor/__init__.py in <module>
----> 1 from .tridescloussortingextractor import TridesclousSortingExtractor

~/anaconda/envs/visification/lib/python3.7/site-packages/spikeextractors/extractors/tridescloussortingextractor/tridescloussortingextractor.py in <module>
      4 
      5 try:
----> 6     import tridesclous as tdc
      7 
      8     HAVE_TDC = True

~/anaconda/envs/visification/lib/python3.7/site-packages/tridesclous/__init__.py in <module>
     25 from .dataio import DataIO
     26 from .signalpreprocessor import offline_signal_preprocessor, estimate_medians_mads_after_preprocesing
---> 27 from .catalogueconstructor import CatalogueConstructor
     28 from .cataloguetools import apply_all_catalogue_steps
     29 from .peeler import Peeler

~/anaconda/envs/visification/lib/python3.7/site-packages/tridesclous/catalogueconstructor.py in <module>
     29 from . import  peakdetector
     30 from . import decomposition
---> 31 from . import cluster
     32 from . import metrics
     33 

~/anaconda/envs/visification/lib/python3.7/site-packages/tridesclous/cluster.py in <module>
     19 
     20 from .sawchaincut import SawChainCut
---> 21 from .pruningshears import PruningShears
     22 
     23 #~ import pyclustering

~/anaconda/envs/visification/lib/python3.7/site-packages/tridesclous/pruningshears.py in <module>
     25 
     26 
---> 27 import hdbscan
     28 
     29 try:

~/anaconda/envs/visification/lib/python3.7/site-packages/hdbscan/__init__.py in <module>
----> 1 from .hdbscan_ import HDBSCAN, hdbscan
      2 from .robust_single_linkage_ import RobustSingleLinkage, robust_single_linkage
      3 from .validity import validity_index
      4 from .prediction import (approximate_predict,
      5                          membership_vector,

~/anaconda/envs/visification/lib/python3.7/site-packages/hdbscan/hdbscan_.py in <module>
     19 from scipy.sparse import csgraph
     20 
---> 21 from ._hdbscan_linkage import (single_linkage,
     22                                mst_linkage_core,
     23                                mst_linkage_core_vector,

hdbscan/_hdbscan_linkage.pyx in init hdbscan._hdbscan_linkage()

ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 88 from C header, got 80 from PyObject
```

What might be causing this?

samuelgarcia, 2021-10-04:
I think you have a hdbscan installation error which cause an error when you import tridesclous.
There is also a probem with montainsort.

I think you are still using the "old" API with spikeextractor/spikesorter/...
You should go for spikeinetrface new release with unique package.

See this : https://spikeinterface.readthedocs.io/en/latest/releases/0.90.0.html

alejoe91, 2021-10-04:
@samuelgarcia it's the new API (`~/anaconda/envs/visification/lib/python3.7/site-packages/spikeinterface/sorters/sorterlist.py`)

samuelgarcia, 2021-10-04:
Yes but there is also `spikeextractors` in the trace, no ?

alejoe91, 2021-10-04:
because of `mountainsort4` (which uses the old API...). The problem is hdbscan though, no?

samuelgarcia, 2021-10-04:
Oh yes. It is an internal call sorry.

Yes the problem is hdbscan.

marcoistasy, 2021-10-08:
that did the trick! thank you!