SpikeInterface/spikeinterface #292: `tridesclous` and `wave_clus` spike sorters not showing as installed

marcoistasy, 2021-10-08:
Hello,

I have installed `tridesclous` and `wave_clus` as per the instructions [here](https://spikeinterface.readthedocs.io/en/latest/install_sorters.html) and have made sure to update the `wave_clus` path by using `WaveClusSorter.set_waveclus_path()`. 

However, when running `ss.installed_sorters()`, I do not see either `tridesclous` or `wave_clus`. An image of my environment as well as the problem is attached below. 

[image]

samuelgarcia, 2021-10-08:
I guess `import tridesclous` is working no ?

marcoistasy, 2021-10-08:
sorry! i meant `tridesclous` and i've updated the issue accordingly.

samuelgarcia, 2021-10-08:
What happen in ipython when you do `import tridesclous as tdc`
Do you have hdbscan ?

alejoe91, 2021-10-14:
@marcoistasy I think that `pip install loky` should fix it!