SpikeInterface/spikeinterface #311: Problem when running `se.NumpyRecording.dump()`

marcoistasy, 2021-10-14:
Hello!

I am running into the follwing `AssertionErrorr` when running `se.NumpyRecording.dump(/path/to.json)`:

```
AssertionError                            Traceback (most recent call last)
~/d/n2b/n2b-visification/scripts/io.py in <module>
     25 # save RecordingExtractor
     26 print('Saving recording to {}'.format(OUTPUT_PATH))
---> 27 recording.dump(OUTPUT_PATH)
     28 print('Finished saving recording.')

~/anaconda/envs/visification/lib/python3.7/site-packages/spikeinterface/core/base.py in dump(self, file_path, relative_to, folder_metadata)
    394         """
    395         if str(file_path).endswith('.json'):
--> 396             self.dump_to_json(file_path, relative_to=relative_to, folder_metadata=folder_metadata)
    397         elif str(file_path).endswith('.pkl') or str(file_path).endswith('.pickle'):
    398             self.dump_to_pickle(file_path, relative_to=relative_to, folder_metadata=folder_metadata)

~/anaconda/envs/visification/lib/python3.7/site-packages/spikeinterface/core/base.py in dump_to_json(self, file_path, relative_to, folder_metadata)
    412             If not None, file_paths are serialized relative to this path
    413         '''
--> 414         assert self.check_if_dumpable()
    415         dump_dict = self.to_dict(include_annotations=True,
    416                                  include_properties=False,

AssertionError: 
```

However, the final `AssertionError` is not printing anything, so I am not quite sure what the issue is. The `OUTPUT_PATH` is `161.json`, by the way.

samuelgarcia, 2021-10-14:
NumpyRecording is not dumpable because it is a in memory object.
But you can save it 
`NumpyRecording.save(folder='/path/to/save')`
`dump()` is an a alis for the `to_dict()` which callect kwargs to reconstrust the object but for numpy array it is not possible.

samuelgarcia, 2021-10-14:
Yes.
```python
cached = numpy_recording.save(folder='your-folder')
# and then later
cached2 = load_extractor('your-folder')
```

cached and cached2 are totaly equivalent and  are lazy.

alejoe91, 2021-10-14:
@marcoistasy don't you have also a `cached.json` file in the folder?