SpikeInterface/spikeinterface #342: get_probe

paul-aparicio, 2021-11-03:
Hello,

I am trying to run a sorter (following the most recent tutorial with my own data), but I get an error saying that I need to call get_probes or get_probegroup

```
# run spike sorting on entire recording
sorting_HS = ss.run_herdingspikes(recording_saved, output_folder=dpath+'/results_HS',
                                  verbose=True, raise_error=True, filter=False)
```
```
---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
/tmp/ipykernel_25625/4201596974.py in <module>
      1 # run spike sorting on entire recording
----> 2 sorting_HS = ss.run_herdingspikes(recording_saved, output_folder=dpath+'/results_HS',
      3                                   remove_existing_folder=True, verbose=True,
      4                                   raise_error=True, filter=False)
      5 print('Found', len(sorting_HS.get_unit_ids()), 'units')

~/anaconda3/envs/sin/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py in run_herdingspikes(*args, **kwargs)
    374 
    375 def run_herdingspikes(*args, **kwargs):
--> 376     return run_sorter('herdingspikes', *args, **kwargs)
    377 
    378 

~/anaconda3/envs/sin/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, with_output, **sorter_params)
     56 
     57     if docker_image is None:
---> 58         sorting = run_sorter_local(sorter_name, recording, output_folder=output_folder,
     59                                    remove_existing_folder=remove_existing_folder,
     60                                    delete_output_folder=delete_output_folder,

~/anaconda3/envs/sin/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py in run_sorter_local(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, **sorter_params)
     78 
     79     # only classmethod call not instance (stateless at instance level but state is in folder)
---> 80     output_folder = SorterClass.initialize_folder(recording, output_folder, verbose, remove_existing_folder)
     81     SorterClass.set_params_to_folder(recording, output_folder, sorter_params, verbose)
     82     SorterClass.setup_recording(recording, output_folder, verbose=verbose)

~/anaconda3/envs/sin/lib/python3.9/site-packages/spikeinterface/sorters/basesorter.py in initialize_folder(cls, recording, output_folder, verbose, remove_existing_folder)
     84 
     85         if cls.requires_locations:
---> 86             locations = recording.get_channel_locations()
     87             if locations is None:
     88                 raise RuntimeError("Channel locations are required for this spike sorter. "

~/anaconda3/envs/sin/lib/python3.9/site-packages/spikeinterface/core/baserecording.py in get_channel_locations(self, channel_ids, locations_2d)
    360         channel_indices = self.ids_to_indices(channel_ids)
    361         if self.get_property('contact_vector') is not None:
--> 362             probe = self.get_probe()
    363             return probe.contact_positions[channel_indices]
    364         else:

~/anaconda3/envs/sin/lib/python3.9/site-packages/spikeinterface/core/baserecording.py in get_probe(self)
    314     def get_probe(self):
    315         probes = self.get_probes()
--> 316         assert len(probes) == 1, 'there are several probe use .get_probes() or get_probegroup()'
    317         return probes[0]
    318 

AssertionError: there are several probe use .get_probes() or get_probegroup()
```
I am not sure how to specify to the sorter that the probe attached to my recording is a probegroup? Or maybe the problem is that I have defined my probe poorly? I am using a 96 channel utah array.

Thanks for any help!

alejoe91, 2021-11-03:
Hi @paul-aparicio 

This is definitely something we should fix. Do you mind sending me your probegroup in probeinterface json format?

You can export it as follows:

```
import probeinterface as pi

probegroup = recording_saved.get_probe_group()
pi.write_probeinterface('my_probe_group.json', probegroup)
```

I'll try to reproduce the issue and provide a fix in the next days :)

Cheers
Alessio

alejoe91, 2021-11-03:
@paul-aparicio no need for the json :) can you try this patch? 

https://github.com/SpikeInterface/spikeinterface/pull/344

alejoe91, 2021-11-03:
Note that for your probes (stereotrode I assume), herdingspikes is not the best option, as it uses an estimate of spike localization in the process (which is hard and probably highly sub-optimal from 2 electrodes)

paul-aparicio, 2021-11-03:
The probe is a 96 channel utah array. Maybe the probe info is incorrect?

closed 2021-11-11 by paul-aparicio