SpikeInterface/spikeinterface #345: Extracting amplitudes from an agreement sorting

ClaraBesserer, 2021-11-04:
Hello,

I want to extract the waveforms and amplitudes of the spikes after performing a multisorting comparison. The waveform extraction works but the amplitude extraction does not. Here is the error message : 

`  File "/home/shiva/Spikeinterface_clara/Ecube_SpikeInterface.py", line 551, in <module>
    main()

  File "/home/shiva/Spikeinterface_clara/Ecube_SpikeInterface.py", line 526, in main
    wf_folder, qc, template_metrics = post_processing(recording_cmr, agreement_sorting, output_folder)

  File "/home/shiva/Spikeinterface_clara/Ecube_SpikeInterface.py", line 365, in post_processing
    amplitudes = st.get_spike_amplitudes(we, outputs="concatenated", progress_bar=True, n_jobs=1, chunk_memory='100M')

  File "/home/shiva/Envs/spikeinterface/spikeinterface/spikeinterface/toolkit/postprocessing/spike_amplitudes.py", line 64, in get_spike_amplitudes
    out = processor.run()

  File "/home/shiva/Envs/spikeinterface/spikeinterface/spikeinterface/core/job_tools.py", line 227, in run
    worker_ctx = self.init_func(*self.init_args)

  File "/home/shiva/Envs/spikeinterface/spikeinterface/spikeinterface/toolkit/postprocessing/spike_amplitudes.py", line 99, in _init_worker_spike_amplitudes
    sorting = load_extractor(sorting)

  File "/home/shiva/Envs/spikeinterface/spikeinterface/spikeinterface/core/base.py", line 758, in load_extractor
    return BaseExtractor.from_dict(file_or_folder_or_dict, base_folder=base_folder)

  File "/home/shiva/Envs/spikeinterface/spikeinterface/spikeinterface/core/base.py", line 303, in from_dict
    extractor = _load_extractor_from_dict(d)

  File "/home/shiva/Envs/spikeinterface/spikeinterface/spikeinterface/core/base.py", line 705, in _load_extractor_from_dict
    extractor = cls(**kwargs)

TypeError: __init__() missing 2 required positional arguments: 'sampling_frequency' and 'multisortingcomparison'`

Also I have an additional question : Why is the multi-jobs not supported for the waveform and amplitude extractions for sortings coming from the AgreementSortingExtractor? I get "BrokenProcessPool: A process in the process pool was terminated abruptly while the future was running or pending" 

Thanks !

alejoe91, 2021-11-04:
Hi, thanks for spotting this bug! I'll fix it :)

What you can do for now is to save the sorting object using the `save` function and proceed with the analysis using the new object:

```
sorting_saved = sorting_agreement.save(folder="sorting_agreement")

# extract waveforms + amplitudes using sorting_saved
```

Let me know if this fixes your proplem while we fix the bug :)

Cheers
Alessio

ClaraBesserer, 2021-11-04:
Hi, yes it's working. Thanks a lot!

samuelgarcia, 2021-11-08:
@alejoe91 : it is not a bug agreement sorting is too complicated to be dumpable.
We should raise an error we we try to dump it.

alejoe91, 2021-11-08:
I agree, we should keep `is_dumpable=False` and check that before dumping

closed 2021-12-25 by alejoe91