SpikeInterface/spikeinterface #346: waveform comparison

khl02007, 2021-11-06:
@alejoe91 @samuelgarcia I'm wondering if there should be a way to compare two waveform extractors (similar in spirit to what we currently have, i.e. the comparison of two sorting extractors based on spike times). For example, I may have two recordings from two different days of a chronically implanted animal. I may have sorted each recording separately for whatever reason (e.g. each recording is too large and cannot be sorted by concatenation; or I may suspect significant drift in the recordings and feel that each should be sorted separately, etc). Then I may want to 'match' a cell from recording 1 to a cell from recording 2 based on waveform alone (spike timing obviously cannot be used since the two recordings are separated in time). 

One simple method could be based on cosine similarity between waveforms of cells in the two recordings. In short, compute pairwise cosine similarity matrix, find the best match, cross off the row and column corresponding to this match, and continue. This method worked reasonable well in my experience (see fig 5 of [this paper](https://www.nature.com/articles/s41467-021-25443-4)). 

Let me know what you think. Happy to discuss more (e.g. what should spikeinterface output be?) and willing to submit a PR or help with implementation.

alejoe91, 2021-11-07:
@khl02007 I think this would be a great addition! I have some code laying around that does exactly that also over multiple sessions (it internally uses the same graph approach as the multi sorting comparison). I'll be happy to discuss and to provide the existing code :) Let's discuss via email

vincisLab, 2021-12-15:
Hello, 

we are interested in doing exactly the same thing discussed here (comparing waveforms on different days of chronically implanted mice).
I  have been wondering why is the status of this addition.
Thanks for all the great work!

alejoe91, 2022-01-13:
Implemented here: https://github.com/SpikeInterface/spikeinterface/pull/366

closed 2022-01-13 by alejoe91