SpikeInterface/spikeinterface #347: Output from export_to_phy() lack channel NPY files

YadongD, 2021-11-08:
Hi,

I am trying to use mountainsort4 to prepossess 64 channels tetrode data and trying to output the data to PHY.
The original data is in binary format, so when I do se.read_openephys(), there is no probe information, then I create Probe by using the following code:

```
probe = ProbeGroup()
for i in range(16):
    tetrode = generate_tetrode()
    tetrode.move([i * 50, 0])
    probe.add_probe(tetrode)
probe.set_global_device_channel_indices(np.arange(64))
```
Then I use mountainsort4 to process the data and try to output the data into PHY by using the following code:

```
export_to_phy(we_MS,sorting_MS, '/content/drive/My Drive/Yadong//phy_folder_for_MS',
              compute_pc_features=False, compute_amplitudes=True)
```

After running, errors as follow show up:

> write_binary_recording with n_jobs 1  chunk_size None
> ---------------------------------------------------------------------------
> AssertionError                            Traceback (most recent call last)
> <ipython-input-81-021437fc317c> in <module>()
>       1 from spikeinterface.exporters import export_to_phy
>       2 
> ----> 3 export_to_phy(we_MS, '/content/drive/My Drive/Yadong//phy_folder_for_MS')
> 
> 2 frames
> /usr/local/lib/python3.7/dist-packages/spikeinterface/exporters/to_phy.py in export_to_phy(waveform_extractor, output_folder, compute_pc_features, compute_amplitudes, sparsity_dict, copy_binary, max_channels_per_template, remove_if_exists, peak_sign, template_mode, dtype, verbose, **job_kwargs)
>     166     channel_maps = np.arange(num_chans, dtype='int32')
>     167     channel_map_si = waveform_extractor.recording.get_channel_ids()
> --> 168     channel_positions = recording.get_channel_locations().astype('float32')
>     169     channel_groups = recording.get_channel_groups()
>     170     if channel_groups is None:
> 
> /usr/local/lib/python3.7/dist-packages/spikeinterface/core/baserecording.py in get_channel_locations(self, channel_ids, locations_2d)
>     360 
>     361     def clear_channel_locations(self, channel_ids=None):
> --> 362         if channel_ids is None:
>     363             n = self.get_num_channel()
>     364         else:
> 
> /usr/local/lib/python3.7/dist-packages/spikeinterface/core/baserecording.py in get_probe(self)
>     314         arr = self.get_property('contact_vector')
>     315         if arr is None:
> --> 316             positions = self.get_property('location')
>     317             if positions is None:
>     318                 raise ValueError('There is not Probe attached to recording. use set_probe(...)')
> 
> AssertionError: there are several probe use .get_probes() or get_probegroup()

The output lacks several channel-related fines, like channel_map.npy.

How could I solve this? Thank you!

alejoe91, 2021-11-08:
Hi @YadongD 

How are you setting the probe to the recording?

YadongD, 2021-11-08:
> Hi @YadongD
> 
> How are you setting the probe to the recording?

Hi Alessio,

Here are the codes(with output from the Colab) that I used before export_to_phy() function. 

`recording = se.read_openephys(folder_path=path)`

> /content/drive/My Drive/Yadong/openEphys/M9/M9_2021-11-03_11-10-22_Record2/Record Node 106/experiment1/recording1/continuous/Rhythm_FPGA-100.0/continuous.dat

```
from probeinterface import generate_tetrode
probe = ProbeGroup()
for i in range(16):
    tetrode = generate_tetrode()
    tetrode.move([i * 50, 0])
    probe.add_probe(tetrode)
probe.set_global_device_channel_indices(np.arange(64))
probe
```

> <probeinterface.probegroup.ProbeGroup at 0x7fbe85069810>

`recording = recording.set_probes(probe)`

> /content/drive/My Drive/Yadong/openEphys/M9/M9_2021-11-03_11-10-22_Record2/Record Node 106/experiment1/recording1/continuous/Rhythm_FPGA-100.0/continuous.dat

```
recording_cmr = recording
recording_f = st.bandpass_filter(recording, freq_min=300, freq_max=6000)
print(recording_f)
recording_cmr = st.common_reference(recording_f, reference='global', operator='median')
print(recording_cmr)
```

> BandpassFilterRecording: 64 channels - 1 segments - 30.0kHz - 603.145s
> CommonReferenceRecording: 64 channels - 1 segments - 30.0kHz - 603.145s

`recording_preprocessed = recording_cmr.save(format='binary')`

> Use cache_folder=/tmp/spikeinterface_cache/tmpkyohfll6/0NL351T7
> write_binary_recording with n_jobs 1  chunk_size None
> /content/drive/My Drive/Yadong/openEphys/M9/M9_2021-11-03_11-10-22_Record2/Record Node 106/experiment1/recording1/continuous/Rhythm_FPGA-100.0/continuous.dat

`sorting_MS = ss.run_mountainsort4(recording=recording_preprocessed)`

> Mountainsort4 use the OLD spikeextractors mapped with RecordingExtractorOldAPI

```
we_MS = si.WaveformExtractor.create(recording_preprocessed, sorting_MS, 'waveforms', remove_if_exists=True)
we_MS.set_params(ms_before=3., ms_after=4., max_spikes_per_unit=500)
we_MS.run_extract_waveforms(n_jobs=1, chunk_size=30000, progress_bar=True)
print(we_MS)
```

> extract waveforms: 100%|##########| 604/604 [00:15<00:00, 39.91it/s]WaveformExtractor: 64 channels - 95 units - 1 segments
>   before:90 after120 n_per_units: 500

```
unit_id0 = sorting_TDC.unit_ids[0]
wavefroms = we_TDC.get_waveforms(unit_id0)
print(wavefroms.shape)
```

> (500, 210, 64)

alejoe91, 2021-11-11:
@YadongD can you try using this PR https://github.com/SpikeInterface/spikeinterface/pull/344?

closed 2022-01-07 by alejoe91