SpikeInterface/spikeinterface #355: Mountainsort4 use the OLD spikeextractors mapped with RecordingExtractorOldAPI

Yuhao-Zhang-SV, 2021-11-10:
When I run _ss.mountainsort4(cached_recording, base_folder)_, where _cached_recording_ is the _**BinaryRecordingExtractor**_ class which is **saved** first and then **load_extract()** from _cache.json_ file, and _base_folder_ is the directory of **_traces_cached_seg.raw_** file, the same as the directory of _cached.json_. However, I got the prompt "**Mountainsort4 use the OLD spikeextractors mapped with RecordingExtractorOldAPI**". So where am I wrong? And how should I call the function ss.mountainsort4()? Thanks in advance~
[image]

alejoe91, 2021-11-10:
Hi, you are not doing anything wrong :) Mountainsort4 uses the old API, so internally we print this message to remind ourselves that we need to update it.

Regarding the error, I think it's because MS4 requires locations and it doesn't seem to find it in your recording object. Did you set a probe?

Yuhao-Zhang-SV, 2021-11-10:
OK, thanks very much!! And I think I have not set a probe yet... Sorry about that. Anyway, thank you very much!

alejoe91, 2021-11-11:
No problem! I'll close the issue for now but feel free to reopen it or open a new one if you find other problems ;)

closed 2021-11-11 by alejoe91