SpikeInterface/spikeinterface #357: trisdesclous channel group error

paul-aparicio, 2021-11-11:
Hello, I am trying to sort data from a 96 channel utah array. I run 
```
sorting_TDC = ss.run_tridesclous(recording_saved, output_folder=dpath+'/results_tdc', 
                                 verbose=True, raise_error=True,
                                 **params)
```
It sorts all of the channels but crashes when trying to save (I think) the results:

```
---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
/tmp/ipykernel_17615/2696511237.py in <module>
      1 # run spike sorting on entire recording
      2 #ss.tridesclous.TridesclousSorter.set_params(wf_left_ms=-1.5, wf_right_ms=2.5)
----> 3 sorting_TDC = ss.run_tridesclous(recording_saved, output_folder=dpath+'/results_tdc', 
      4                                  verbose=True, raise_error=True,
      5                                  **params)

~/anaconda3/envs/sin/lib/python3.9/site-packages/spikeinterface-0.91.1.dev0-py3.9.egg/spikeinterface/sorters/runsorter.py in run_tridesclous(*args, **kwargs)
    327 
    328 def run_tridesclous(*args, **kwargs):
--> 329     return run_sorter('tridesclous', *args, **kwargs)
    330 
    331 

~/anaconda3/envs/sin/lib/python3.9/site-packages/spikeinterface-0.91.1.dev0-py3.9.egg/spikeinterface/sorters/runsorter.py in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, with_output, **sorter_params)
     56 
     57     if docker_image is None:
---> 58         sorting = run_sorter_local(sorter_name, recording, output_folder=output_folder,
     59                                    remove_existing_folder=remove_existing_folder,
     60                                    delete_output_folder=delete_output_folder,

~/anaconda3/envs/sin/lib/python3.9/site-packages/spikeinterface-0.91.1.dev0-py3.9.egg/spikeinterface/sorters/runsorter.py in run_sorter_local(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, **sorter_params)
     87     SorterClass.run_from_folder(output_folder, raise_error, verbose)
     88     if with_output:
---> 89         sorting = SorterClass.get_result_from_folder(output_folder)
     90     else:
     91         sorting = None

~/anaconda3/envs/sin/lib/python3.9/site-packages/spikeinterface-0.91.1.dev0-py3.9.egg/spikeinterface/sorters/basesorter.py in get_result_from_folder(cls, output_folder)
    252                 "Spike sorting failed. You can inspect the runtime trace in spikeinterface_log.json")
    253 
--> 254         sorting = cls._get_result_from_folder(output_folder)
    255         return sorting
    256 

~/anaconda3/envs/sin/lib/python3.9/site-packages/spikeinterface-0.91.1.dev0-py3.9.egg/spikeinterface/sorters/tridesclous/tridesclous.py in _get_result_from_folder(cls, output_folder)
    167     @classmethod
    168     def _get_result_from_folder(cls, output_folder):
--> 169         sorting = TridesclousSortingExtractor(folder_path=output_folder)
    170         return sorting
    171 

~/anaconda3/envs/sin/lib/python3.9/site-packages/spikeinterface-0.91.1.dev0-py3.9.egg/spikeinterface/extractors/tridesclousextractors.py in __init__(self, folder_path, chan_grp)
     24             # if chan_grp is not provided, take the first one if unique
     25             chan_grps = list(dataio.channel_groups.keys())
---> 26             assert len(chan_grps) == 1, 'There are several groups in the folder, specify chan_grp=...'
     27             chan_grp = chan_grps[0]
     28 

AssertionError: There are several groups in the folder, specify chan_grp=...
```

quarying my recording object for the channel groups shows that they are there?:
```
[60]: recording_prb.get_channel_groups()

[61]: array([ 0,  1,  2,  3,  4,  5,  6,  7,  8,  9, 10, 11, 12, 13, 14, 15, 16,
       17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33,
       34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49, 50,
       51, 52, 53, 54, 55, 56, 57, 58, 59, 60, 61, 62, 63, 64, 65, 66, 67,
       68, 69, 70, 71, 72, 73, 74, 75, 76, 77, 78, 79, 80, 81, 82, 83, 84,
       85, 86, 87, 88, 89, 90, 91, 92, 93, 94, 95])
```
Thank you for any help

samuelgarcia, 2021-11-12:
Are you using an already existing folder that contain sorting with group at tridesclous level (and not spikeinterface level) ?

paul-aparicio, 2021-11-12:
No, I don't believe so. The output folder I specify in the call to ss.run_trisdesclous is unique and doesn't exist prior to running the sorter?

samuelgarcia, 2021-11-12:
how did you set grouping ? Is it one group per channel or one global group ?
Are you using a PRB file that contain groups ?

paul-aparicio, 2021-11-12:
The grouping information comes from a probe file (.prb). Its a probegroup where each channel is its own group.
```
# load probe information
probe = pi.read_prb(probe_file)
probe.to_dataframe(complete=True)

  | probe_index | x | y | contact_shapes | radius | shank_ids | contact_ids | device_channel_indices | si_units | plane_axis_x_0 | plane_axis_x_1 | plane_axis_y_0 | plane_axis_y_1
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
0 | 3600.0 | 400.0 | circle | 5.0 |   |   | 0 | um | 1.0 | 0.0 | 0.0 | 1.0
1 | 3600.0 | 800.0 | circle | 5.0 |   |   | 1 | um | 1.0 | 0.0 | 0.0 | 1.0
2 | 3600.0 | 1200.0 | circle | 5.0 |   |   | 2 | um | 1.0 | 0.0 | 0.0 | 1.0
3 | 3600.0 | 1600.0 | circle | 5.0 |   |   | 3 | um | 1.0 | 0.0 | 0.0 | 1.0
4 | 3600.0 | 2000.0 | circle | 5.0 |   |   | 4 | um | 1.0 | 0.0 | 0.0 | 1.0
... | ... | ... | ... | ... | ... | ... | ... | ... | ... | ... | ... | ...
91 | 0.0 | 1600.0 | circle | 5.0 |   |   | 91 | um | 1.0 | 0.0 | 0.0 | 1.0
92 | 0.0 | 2000.0 | circle | 5.0 |   |   | 92 | um | 1.0 | 0.0 | 0.0 | 1.0
93 | 0.0 | 2400.0 | circle | 5.0 |   |   | 93 | um | 1.0 | 0.0 | 0.0 | 1.0
94 | 0.0 | 2800.0 | circle | 5.0 |   |   | 94 | um | 1.0 | 0.0 | 0.0 | 1.0
95 | 0.0 | 3200.0 | circle | 5.0 |   |   | 95 | um | 1.0 | 0.0 | 0.0 | 1.0

<p>96 rows × 13 columns</p>

```

paul-aparicio, 2021-11-16:
Hello @samuelgarcia,

from the pandas output 'shank_ids' and 'contact_ids' are blank, while 'device_channel_indices' have values. Is that ok? Below is the first few lines of the probe file being used if that is helpful. Thanks for any help!

```
total_nb_channels = 96
radius = 100

channel_groups = {
	# Shank index
	0:
		{
		'channels': [0],
		'geometry': {
			0: [3600.0, 400.0],
			}
		},
	1:
		{
		'channels': [1],
		'geometry': {
			1: [3600.0, 800.0],
			}
		},
	2:
		{
		'channels': [2],
		'geometry': {
			2: [3600.0, 1200.0],
			}
		},
	3:
		{
		'channels': [3],
		'geometry': {
			3: [3600.0, 1600.0],
			}
		},

            ...

```

alejoe91, 2021-11-16:
@paul-aparicio you should use the `run_sorter_by_property()` function! That's splits in groups before spike sorting

paul-aparicio, 2021-11-18:
Thank you! This seems to work, though I still don't understand why the sorter crashes when trying to finish up after performing all of the sorting when using ss.run_tridesclous?

closed 2022-01-07 by alejoe91