SpikeInterface/spikeinterface #360: Ironclust Error Message

vincisLab, 2021-11-12:
Hi, we were able to successfully run Ironclust through the old version of spikeinterface on a Mac. However, trying to run Ironclust using the NEW spikeinterface on Windows on the same dataset as before produces this error. Interestingly, it seems like the sorting ran successfully.

Clustering result wrote to D:\ironclust_test_with_params2\tmp/firings.mda

#SF-SORTER-RUNTIME#110.460#

Error using <a href="matlab:matlab.internal.language.introspective.errorDocCallback('quit')" style="font-weight:bold">quit</a>

Unrecognized function or variable 'Settings'.

Error in <a href="matlab:matlab.internal.language.introspective.errorDocCallback('run_ironclust', 'D:\ironclust_test_with_params2\tmp\run_ironclust.m', 10)" style="font-weight:bold">run_ironclust</a> (<a href="matlab: opentoline('D:\ironclust_test_with_params2\tmp\run_ironclust.m',10,0)">line 10</a>)

quit(0);

 

ironclust run time 122.03s

Any information you can give on why this error appears would be greatly appreciated!

samuelgarcia, 2021-11-15:
Hi @vincisLab 
A file `run_ironclust.m` has been generated in your folder. Could you run it from matlab directly and check the error message ?

vincisLab, 2021-11-17:
Hi @samuelgarcia ,

Matlab does not supply an error message. This is the result of running run_ironclust.m in MATLAB (stopped with a breakpoint right before quitting):

>> run_ironclust
===================================================
IronClust Version: 2
===================================================
Removed 0 lock(s).
Running irc2.m (5.9.8)
Created D:\ironclust_test_with_params2\tmp\raw_geom.prm
Loaded from cache: D:\ironclust_test_with_params2\tmp\detect_ad047dc452190324db16a4d8506d3f57\detect_irc.mat
Loaded from cache: D:\ironclust_test_with_params2\tmp\detect_ad047dc452190324db16a4d8506d3f57\sort_b5e8be3adbd834d6471ca94e5d41d629\sort_irc.mat
Loaded from cache: D:\ironclust_test_with_params2\tmp\detect_ad047dc452190324db16a4d8506d3f57\sort_b5e8be3adbd834d6471ca94e5d41d629\auto_f83832cfc7a58d1733a09ba4456e6515\auto_irc.mat

------------------------------
Summary of D:\ironclust_test_with_params2\tmp\raw_geom.prm
------------------------------
Recording format
    Recording file:         D:\ironclust_test_with_params2\ironclust_dataset\raw.mda
    Probe file:             D:\ironclust_test_with_params2\ironclust_dataset\geom.csv
    Recording Duration:     3518.6s
    Data Type:              single
    #Channels in file:      32
    #Sites:                 32
    #Shanks:                1
Pre-processing
    Filter type:            bandpass
    Filter range (Hz):      [NaN, NaN]
    Matched Filter:         0
    Common ref:             whiten-trimmean
    Whiten:                 0
    FFT threshold:          8
    blank threshold:        0
Events
    #Spikes:                142454
    Feature extracted:      gpca
    #Sites/event:           16
    maxDist_site_um:        50
    maxDist_site_spk_um:    150
    spkLim_ms:              [-0.250, 0.750]
    #PC/chan:               9
Cluster
    #Clusters:              20
    #Unique events:         137217
    min. spk/clu:           30
    Cluster method:         drift-knn
    knn:                    30
    step_sec_drift:         20.0s
    batch_sec_drift:        300.0s
Auto-merge
    merge_overlap_thresh:   0.950
    delta_cut:              1.000
    merge_thresh_cc:        1.000
    maxWavCor:              0.985
Runtime (s)
    Detect + feature (s):   89.3s
    Cluster runtime (s):    9.4s
    merge runtime (s):      7.6s
    Total runtime (s):      106.3s
    Runtime speed:          x33.1 realtime
    Processing speed:       1340.2 spikes/s
memory usage (GiB):         0.201
    detect:                 0.014
    sort:                   0.115
    auto-merge:             0.201
Execution
    irc2 version:           5.9.8
    fGpu (GPU use):         0
    fParfor (parfor use):   0
    fLargeRecording:        0
    Parameter file:         D:\ironclust_test_with_params2\tmp\raw_geom.prm
------------------------------
Clustering result wrote to D:\ironclust_test_with_params2\tmp/firings.mda
#SF-SORTER-RUNTIME#2.691#
10  quit(0);

 spikeinterface gives the following output:

Error using <a href="matlab:matlab.internal.language.introspective.errorDocCallback('quit')" style="font-weight:bold">quit</a>

Unrecognized function or variable 'Settings'.

Error in <a href="matlab:matlab.internal.language.introspective.errorDocCallback('run_ironclust', 'D:\spikesorting\dataTest\CB250\CB250_082421_recording\results_ironclust\tmp\run_ironclust.m', 10)" style="font-weight:bold">run_ironclust</a> (<a href="matlab: opentoline('D:\spikesorting\dataTest\CB250\CB250_082421_recording\results_ironclust\tmp\run_ironclust.m',10,0)">line 10</a>)

quit(0);

We are using a Windows computer with MATLAB 2021b. Running the same code on a Mac with an older version of MATLAB worked.

vincisLab, 2021-11-17:
Hi @samuelgarcia ,

Matlab does not supply an error message. This is the result of running run_ironclust.m in MATLAB (stopped with a breakpoint right before quitting):

>> run_ironclust
===================================================
IronClust Version: 2
===================================================
Removed 0 lock(s).
Running irc2.m (5.9.8)
Created D:\spikesorting\dataTest\CB250\CB250_082421_recording\results_ironclust\tmp\raw_geom.prm
Loaded from cache: D:\spikesorting\dataTest\CB250\CB250_082421_recording\results_ironclust\tmp\detect_ec570d7a8d8601f71c0bd4ca9167c38b\detect_irc.mat
Loaded from cache: D:\spikesorting\dataTest\CB250\CB250_082421_recording\results_ironclust\tmp\detect_ec570d7a8d8601f71c0bd4ca9167c38b\sort_b5e8be3adbd834d6471ca94e5d41d629\sort_irc.mat
Loaded from cache: D:\spikesorting\dataTest\CB250\CB250_082421_recording\results_ironclust\tmp\detect_ec570d7a8d8601f71c0bd4ca9167c38b\sort_b5e8be3adbd834d6471ca94e5d41d629\auto_f83832cfc7a58d1733a09ba4456e6515\auto_irc.mat

------------------------------
Summary of D:\spikesorting\dataTest\CB250\CB250_082421_recording\results_ironclust\tmp\raw_geom.prm
------------------------------
Recording format
    Recording file:         D:\spikesorting\dataTest\CB250\CB250_082421_recording\results_ironclust\ironclust_dataset\raw.mda
    Probe file:             D:\spikesorting\dataTest\CB250\CB250_082421_recording\results_ironclust\ironclust_dataset\geom.csv
    Recording Duration:     781.0s
    Data Type:              single
    #Channels in file:      32
    #Sites:                 32
    #Shanks:                1
Pre-processing
    Filter type:            bandpass
    Filter range (Hz):      [300.0, 8000.0]
    Matched Filter:         0
    Common ref:             trimmean
    Whiten:                 0
    FFT threshold:          8
    blank threshold:        0
Events
    #Spikes:                96215
    Feature extracted:      gpca
    #Sites/event:           12
    maxDist_site_um:        50
    maxDist_site_spk_um:    100
    spkLim_ms:              [-0.250, 0.750]
    #PC/chan:               9
Cluster
    #Clusters:              19
    #Unique events:         92789
    min. spk/clu:           30
    Cluster method:         drift-knn
    knn:                    30
    step_sec_drift:         20.0s
    batch_sec_drift:        300.0s
Auto-merge
    merge_overlap_thresh:   0.950
    delta_cut:              1.000
    merge_thresh_cc:        1.000
    maxWavCor:              0.985
Runtime (s)
    Detect + feature (s):   43.0s
    Cluster runtime (s):    4.5s
    merge runtime (s):      3.1s
    Total runtime (s):      50.6s
    Runtime speed:          x15.4 realtime
    Processing speed:       1902.7 spikes/s
memory usage (GiB):         2.573
    detect:                 2.541
    sort:                   2.573
    auto-merge:             2.364
Execution
    irc2 version:           5.9.8
    fGpu (GPU use):         1
    fParfor (parfor use):   0
    fLargeRecording:        0
    Parameter file:         D:\spikesorting\dataTest\CB250\CB250_082421_recording\results_ironclust\tmp\raw_geom.prm
------------------------------
Clustering result wrote to D:\spikesorting\dataTest\CB250\CB250_082421_recording\results_ironclust\tmp/firings.mda
#SF-SORTER-RUNTIME#0.378#
10  quit(0);
spikeinterface gives the following output:

Error using quit

Unrecognized function or variable 'Settings'.

Error in run_ironclust (line 10)

quit(0);

We are using a Windows computer with MATLAB 2021b. Running the same code on a Mac with an older version of MATLAB worked.

vincisLab, 2021-11-17:
@samuelgarcia Despite the error message, Ironclust seems to still work...spikeinterface says sorting returned 20 units.

alejoe91, 2021-11-17:
@vincisLab you can load the spikesorted outout with:

```
import spikeinterface.sorters as ss

sorting_IC = ss.IronClustSorter.get_result_from_folder(output_folder)
```

vincisLab, 2021-11-17:
Hi @alejoe91 , yes I'm able to get the results of the sorting. I suppose I can ignore the warning since it doesn't appear to affect the ouput.

vincisLab, 2021-11-17:
We're also trying to run Kilosort3 and we're getting a similar error message about 'Settings' being an unrecognized variable or function. On top of that message, we're also getting one about subscript indicies:

Error using gpuArray/subsref

Subscript indices must either be real positive integers or logicals.Error using <a href="matlab:matlab.internal.language.introspective.errorDocCallback('quit')" style="font-weight:bold">quit</a>

Unrecognized function or variable 'Settings'.

Error in <a href="matlab:matlab.internal.language.introspective.errorDocCallback('kilosort3_master', 'D:\spikesorting\dataTest\CB250\CB250_082421_recording\results_kilosort\kilosort3_master.m', 43)" style="font-weight:bold">kilosort3_master</a> (<a href="matlab: opentoline('D:\spikesorting\dataTest\CB250\CB250_082421_recording\results_kilosort\kilosort3_master.m',43,0)">line 43</a>)

    quit(1);

 

ERROR: MATLAB error Exit Status: 0x00000001

Error running kilosort3
Traceback (most recent call last):
  File "D:\AnacondaStuff\lib\site-packages\spikeinterface\sorters\basesorter.py", line 198, in run_from_folder
    SorterClass._run_from_folder(output_folder, sorter_params, verbose)
  File "D:\AnacondaStuff\lib\site-packages\spikeinterface\sorters\kilosortbase.py", line 42, in _run_from_folder
    raise Exception(f'{cls.sorter_name} returned a non-zero exit code')
Exception: kilosort3 returned a non-zero exit code

We saw a similar issue at https://github.com/MouseLand/Kilosort/issues/443 and they recommended changing the number of channels in the kilosort GUI. We are unsure if this is the correct fix or how we would do this with spikeinterface.

Thank you for all your help!

samuelgarcia, 2021-11-17:
@vincisLab : for ironclust I don't understand what is happening. It raise an error at python level even if there is no error at matlab level ? Can you inspect the **spikeinterface_log.json** file to try to guess ?

For kilosort at spikeinterface level you can remove channel doing this

```
recording2 = recording.channel_slice(some_channel_ids)
```
and run the sorter on recording2

vincisLab, 2021-11-17:
@samuelgarcia I'm having a difficult time understanding what's going on as well. I believe you are correct. My best guess is that Python is hitting an error while telling matlab to quit, and it crashes matlab (which basically accomplishes what it was trying to do anyways). Here's the output from the log file:

"Clustering result wrote to D:\\spikesorting\\dataTest\\CB250\\CB250_082421_recording\\results_ironclust\\tmp/firings.mda",
        "#SF-SORTER-RUNTIME#58.838#",
        "{\bError using <a href=\"matlab:matlab.internal.language.introspective.errorDocCallback('quit')\" style=\"font-weight:bold\">quit</a>",
        "Unrecognized function or variable 'Settings'.",
        "",
        "Error in <a href=\"matlab:matlab.internal.language.introspective.errorDocCallback('run_ironclust', 'D:\\spikesorting\\dataTest\\CB250\\CB250_082421_recording\\results_ironclust\\tmp\\run_ironclust.m', 10)\" style=\"font-weight:bold\">run_ironclust</a> (<a href=\"matlab: opentoline('D:\\spikesorting\\dataTest\\CB250\\CB250_082421_recording\\results_ironclust\\tmp\\run_ironclust.m',10,0)\">line 10</a>)",
        "quit(0);",
        "}\b"

Sorry for all the confusion.

samuelgarcia, 2021-11-18:
The error message looks weird with this kind of `<a href=`

@alejoe91 : do you remeber why with have `quit(0);` at th end of the generated script ?
Maybe removing it (line 208) could fix the stuff. @vincisLab : could you try this ?

alejoe91, 2021-11-18:
Seems related to this: https://ch.mathworks.com/matlabcentral/answers/1566246-got-error-using-exit-in-nodesktop-mode

@vincisLab can you try to do the suggested action? Disable the "help imporve MATLAB" from the settings

samuelgarcia, 2021-11-18:
Is the quit(0) necessary ?

alejoe91, 2021-11-18:
yes, because that triggers an error on the SI side

samuelgarcia, 2021-11-18:
the quitt(1) trigger an error but why the quit(0) ?

vincisLab, 2021-11-18:
Hi guys, thanks for the help.

@alejoe91 removing the "help improve MATLAB" from the preferences prevent the error to appear, so all good with Ironclust.

@samuelgarcia  As per Kilosort3 we still get the error even changing the number of channel with the channel_slice command .

here is the error:

Time   0s. Computing whitening matrix.. 

Getting channel whitening matrix... 

Channel-whitening matrix computed. 

Time   3s. Loading raw data and applying filters... 

Time   8s. Finished preprocessing 477 batches. 

----------------------------------------Error using gpuArray/subsref

Subscript indices must either be real positive integers or logicals.ERROR: MATLAB error Exit Status: 0x00000001

Error running kilosort3
Traceback (most recent call last):
  File "D:\AnacondaStuff\lib\site-packages\spikeinterface\sorters\basesorter.py", line 198, in run_from_folder
    SorterClass._run_from_folder(output_folder, sorter_params, verbose)
  File "D:\AnacondaStuff\lib\site-packages\spikeinterface\sorters\kilosortbase.py", line 42, in _run_from_folder
    raise Exception(f'{cls.sorter_name} returned a non-zero exit code')
Exception: kilosort3 returned a non-zero exit code

---------------------------------------------------------------------------
SpikeSortingError                         Traceback (most recent call last)
<ipython-input-6-94511fe93d7c> in <module>
----> 1 sorting_IC = ss.run_kilosort3(recording_saved2, verbose=True, 
      2                                 output_folder = spikeInt_folder / 'results_kilosort',
      3                                 n_jobs_bin = 6, detect_threshold = 6)

D:\AnacondaStuff\lib\site-packages\spikeinterface\sorters\runsorter.py in run_kilosort3(*args, **kwargs)
    369 
    370 def run_kilosort3(*args, **kwargs):
--> 371     return run_sorter('kilosort3', *args, **kwargs)
    372 
    373 

D:\AnacondaStuff\lib\site-packages\spikeinterface\sorters\runsorter.py in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, with_output, **sorter_params)
     56 
     57     if docker_image is None:
---> 58         sorting = run_sorter_local(sorter_name, recording, output_folder=output_folder,
     59                                    remove_existing_folder=remove_existing_folder,
     60                                    delete_output_folder=delete_output_folder,

D:\AnacondaStuff\lib\site-packages\spikeinterface\sorters\runsorter.py in run_sorter_local(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, **sorter_params)
     85         recording, output_folder, sorter_params, verbose)
     86     SorterClass.setup_recording(recording, output_folder, verbose=verbose)
---> 87     SorterClass.run_from_folder(output_folder, raise_error, verbose)
     88     if with_output:
     89         sorting = SorterClass.get_result_from_folder(output_folder)

D:\AnacondaStuff\lib\site-packages\spikeinterface\sorters\basesorter.py in run_from_folder(cls, output_folder, raise_error, verbose)
    232         if has_error and raise_error:
    233             print(log['error_trace'])
--> 234             raise SpikeSortingError(
    235                 f"Spike sorting failed. You can inspect the runtime trace in {output_folder}/spikeinterface_log.json")
    236 

Thanks again for all the help!

alejoe91, 2022-04-01:
close for lack of activity

closed 2022-04-01 by alejoe91