SpikeInterface/spikeinterface #361: Issue with PYPI installation instructions on documentation site

TomDonoghue, 2021-11-14:
The install instructions for using PYPI the website currently look as follows:

[image]

I think there are a couple small issues here:
1) to make the 'full' option work from pip, I need to use quotes (`pip install "spikeinterface[full]"`)
     - I'm not too familiar with these pip options / possible terminal quirks, so I'm not sure if this is _always_ required, but this probably could / should be mentioned, so people don't get stuck in cases in which it is needed
2) the second part, about not installing all dependencies repeats the "full" option but presumably should be something different
    - Should this read `pip install spikeinterface`, or maybe something like `pip install "spikeinterface[core]"`?

The source is this [page](https://github.com/SpikeInterface/spikeinterface/blob/master/doc/installation.rst), and I'd be happy to PR a fix, but I'm not 100% sure on what _should_ be written here, so I thought I'd bring it up here first.

alejoe91, 2021-11-14:
Hi @TomDonoghue 

Thanks for spotting this! Indeed the second option should only be `pip install spikeinterface` without anything else.

What's your OS? The `pip install spikeinterface[full]` works well on our side on UNIX systems without quotation marks :)

TomDonoghue, 2021-11-14:
After a quick look into this, I think the need to use quotes is because I'm using Z shell. I'm on a mac, which recently switched to using Z shell, so this will probably be a fairly common situation. 

In #362 I updated the file to address both things - adding a note about quotes for Z shell, and updating the core install line.

TomDonoghue, 2021-11-15:
This was fixed / merged with #362, so I'll close this now!

closed 2021-11-15 by TomDonoghue