SpikeInterface/spikeinterface #363: Preprocessing Questions

ferchaure, 2021-11-15:
Hi guys, small question, what do you think about a similar class than  FilterRecording (and  FilterOpenCLRecording) but with generic inputs like coef (the ba or sos values) instead of the filter description? 

I have an awful recording with a lot of peaks in the spectrum; I made notch filters with custom q factors and I would like to do all of them in one filter instead of a long series of single notch filters. A NotchFilterRecording that handle multiple frequencies will be enough for me but maybe is more useful to have a generic filtering that uses all the fancy stuff of the library.

Let me know what you think about that ideas and I can make a pull request.

alejoe91, 2021-11-15:
Hola @ferchaure 

I think we could have both! A multi-notch filter and a generic filter. Note that both can use already the `FilterRecordingSegment`, which already accepts coeffs and filter type (https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/toolkit/preprocessing/filter.py#L84-L91)

Would you have time to implement it? 

Cheers
Alessio

ferchaure, 2021-11-15:
Hi Alessio, I will do a pull request today and we can see the details there.  

I keep this issue open because I have a few other questions:
- About detect_peaks, what do you think about adding a parameter to remove duplicate detections in all channels... or maybe a function to clean the output of detect_peaks. I was thinking in a really simple implementation of this ideas: [https://doi.org/10.3390/brainsci11060761](https://doi.org/10.3390/brainsci11060761)

alejoe91, 2021-11-15:
> Hi Alessio, I will do a pull request today and we can see the details there.

Thanks :)

> I keep this issue open because I have a few other questions:
>
> * About detect_peaks, what do you think about adding a parameter to remove duplicate detections in all channels... or maybe a function [...]

So the `locally_exclusive` mode should already do somehting similar, but of course we are open to add more options/methods!

samuelgarcia, 2021-11-15:
Hi Fernando
I think It will be easy to modify the `FilterRecording` class to add optional filter coefficient in **ab** or **sos** mode.
Lets discuss in a PR.

And yes `locally_exclusive` already do what you want when a spike is on several channel : it keep the best channel.
You will need to play with `local_radius_um`
Note that there is also the `n_shifts` also remove duplicate on time axis when there is rebound.

More important, consider using this branch which have several improvement on `detect_peak()`: #337

samuelgarcia, 2021-11-15:
I just look in the abstract of the paper.
The "deduplication" will be covered at template matching level which is not implemenbted yet but should be day in a near futur.

ferchaure, 2021-11-18:
I already ask you about this a while back, but maybe now is possible with some modifications.
I would like to mask out segments of signal with high power (mountainlab had something like that, and it was useful sometimes ).
What do you think about this steps:
- Take a recording and get power statistics. The idea is to approx the mean and std of the power in segments of NM samples. Then create a threshold using that values.
- Overwrite  each segment of NM  samples that exceed the threshold with zeros. 
- To do this in a lazy way, maybe I could use a  boolean flag for each segment of NM samples, that is calculated just when get_traces is called for that samples and then saved. In that way the power of the signal segments are just calculated one time each (the flags for the ones used for statistics can be saved).

Does this makes sense? 
Can I update the recording info (that boolean flags) in some calls of get_traces?

samuelgarcia, 2021-11-19:
Hi Fernando,
I think what you propose is similar with `RemoveArtifactsRecording` but trigger would be computed with some statistic.
But ms_before/ms_after would not be fixed.
If you go for this I would recommend to refactor `RemoveArtifactsRecording` and to build your idea with some mixin class or somethig like this.
boolean mask is also a simple solution but it it lead to more memory when multi processing.

Also, I would use median/mad instead of mean/std to have more robust threhold.

ferchaure, 2021-11-19:
Ok I will do a pull request later, the original idea mask out all the channels when any of them exceed the threshold. I could let the code in that way and let the user user split for mask out just groups of channels, or I could add an optional input in a function similar to  remove_artifacts, that split by property, apply the masks  and aggregate the recording. 

I have something like that to apply whiten by property and not all the recording together, do you want a pull request for that as well?

samuelgarcia, 2021-11-19:
You can also, use the mask and then detect up/dowk front in the mask and that will be the "period limits" of artifact.

ferchaure, 2021-11-26:
Hi @samuelgarcia, about peak detection, I have been doing some tests and at least for my recordings removing events when they are detected in all channels of the recording (or maybe a parametrized big proportion of them ) is a fast and reliable way to remove a family of annoying artifacts. This is the Part 1 of the algorithm discuss on the paper above, the other parts are a bit more controversial. 

I don't know which is the best way to implement this criteria in spikeinterface:
1. A separated function that curates peaks from the peak dictionary (the output of _detect_peaks_chunk). This is the easiest way to implement it but I won't help for the 'locally_exclusive' detection because some detections are already filtered. This could be just one possible way to label artifacts, other ones could be implemented later using the waveforms as well. If you like this idea where should I put this funcion?
2. Edit both detection methods (detect_peaks_by_channel and detect_peak_locally_exclusive) and adding them the parameters to remove artifacts. This is the more general alternative... but could be extra complicated to merge this criteria with detect_peak_locally_exclusive.
3. Adding another method, similar to 'by_channel'  but with the criteria to remove the duplicated events. I will need to include extra parameters (time window and proporcion of channels).  Probably it's not the best idea to add more inputs to detect_peaks... just because another detection methods can be created and keep increasing the inputs will be a mess. I don't know the proper way to implement that dynamic input and let the users know the parameters of each method.

I guess that option 1 or 3 are good, which one will depend if you like the idea of a "detection curation" before spike sorting ,  what do you think?

samuelgarcia, 2021-11-26:
Which method did you test ? 'by_channel' or 'locally_exclusive' ?

For me the best would to implement a new function and add a new `method` case the point to this function.
So 3.

In my mind, detection can be done on the fly if possible.

Having also a familly of function to `clean_peak()` is also a good idea but I would do it in a separate sub module.
The idea of `sortingcomponent` is that evryone must be happy with it. So it has to be flexible.

ferchaure, 2021-11-26:
I tested it with by_channel (I don't have neighborhoods of channels). The type of artifact that adds detections in all the channels (in my case I think it was the patient using a cellphone) will affect the 'locally_exclusive' method as well, but just with one detection per neighborhood. 

If I do 3, how should I pass the additional two parameters to detect_peaks?

ferchaure, 2021-11-26:
I'm intrigue about the possible uses of a component to clean peaks, some of them could be:
- The criteria above
-  Max amplitude threshold (old school, but maybe useful) 
- Remove overlapping spikes using a refractory period. If we want to implement something similar to classic spike sorters using `sortingcomponent  ` we need to separate overlapping spikes before feature extraction. 
- Using the waveforms to filter artifacts, Combinato has some of this criteria

What do you think about redefine neighborhoods from correlations in the spike trains? that could be interesting

samuelgarcia, 2021-11-29:
Yes, you can  add as many kwargs you need in detect_peak.

Foe the clean_peak, if there is a real need maybe it is better to start discussion with some drafty code.
I think also here we need flexibility and multi method.

ferchaure, 2021-11-30:
@samuelgarcia doing some test for the new method, I may have found a bug in detect_peaks when the extractor has a gain. 

get_noise_levels() returned the scaled noise level but in `get_chunk_with_margin `(in toolkit\preprocessing\tools.py), the output of  `rec_segment.get_traces` is nos scaled. Therefore the threshold is scaled but compared with the unscale data.

A quick fix could be use `return_scaled=False` in get_noise_levels... but then the amplitude returned in the peaks output will be unscaled. 

let me know if you want a pull request

samuelgarcia, 2021-11-30:
Yes you are right [here](https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/sortingcomponents/peak_detection.py#L80) we need `return_scaled=False`
I generally compute noise level outside the function (with return_scaled=False) this is why I did not trigger the problem.
Thanks.

samuelgarcia, 2021-12-01:
@ferchaure : I did this quickly #389
We will make a release today, can you confirm this is ok now ?

ferchaure, 2021-12-02:
Idea: add parameter by_property to whiten() to apply the whitening by groups and not to all the channels together. (similar to  [this code](https://github.com/ferchaure/spikeinterface/blob/484350a27038ccdfde99565c0f34e75d78aeac88/spikeinterface/toolkit/preprocessing/mask_out_artifacts.py#L122))

Can I do a pull request for that?

alejoe91, 2021-12-03:
Yes that's a good idea @ferchaure !!

ferchaure, 2021-12-03:
all the ideas in their pull request.

closed 2021-12-03 by ferchaure