SpikeInterface/spikeinterface #367: Event-related potentials question

AlanoudAlsubhi, 2021-11-17:
Hi, 

I tried to read the documentation to find an answer for my question but no luck. 
I am trying analyze event-related potentials. Is there a tutorial on that? 
The closest I was able to find was `spikeinterface.core.BaseEvent(channel_ids, structured_dtype)` but there isn't much description about it. 

I have my stimulus added to my NWB file using this 

```
from pynwb import TimeSeries
trigger_ts = TimeSeries(name='trigger_timeseries', data=trigger_data, unit='NA', starting_time=0.0, rate=30000.0)
nwbfile.add_stimulus(trigger_ts)

```
I appreciate any help or any assistance directing me to a helpful source. 
Thanks, 
Alanoud

samuelgarcia, 2021-11-18:
Hi Alamoud,
spikeinterface is oriented for spike sorting task.

It has a small interface to suport event (aka trigger) with BaseEvent.
This event stuff have to be enhenced to be more flexible.

Analysing event related potential is doable with spikeinterface API but there there is nothing ready to use nor example for this.

alejoe91, 2021-11-18:
I would recommend using [`elephant`](http://neuralensemble.org/elephant/) for this kind of analysis

closed 2021-12-25 by alejoe91