SpikeInterface/spikeinterface #370: How to run Mountainsort4?

Yuhao-Zhang-SV, 2021-11-18:
So basically I need to set probe before I ran Mountainsort4, right? Could you please provide / show me a simple example on the pipeline of running Mountainsort4, including saving Recording object, setting probes, etc.? Since I read about the tutorial on how to use run_tridesclous(), but it seems that there are some spikesorting errors... Thanks in advance~

----------------------------------------------------------------------------------------------------

Hi, you are not doing anything wrong :) Mountainsort4 uses the old API, so internally we print this message to remind ourselves that we need to update it.

Regarding the error, I think it's because MS4 requires locations and it doesn't seem to find it in your recording object. Did you set a probe?

_Originally posted by @alejoe91 in https://github.com/SpikeInterface/spikeinterface/issues/355#issuecomment-965749124_

samuelgarcia, 2021-11-18:
It will be exactly the same for all sorters including Mountainsort4.
just use
`sorting = run_sorter('mountainsort4', recording)`
internally there is a wrapper to old API (this is why you have the message) but at end user you have have to worry about it.

Yuhao-Zhang-SV, 2021-11-18:
OK, so do I need to set probe info first before run_sorter()? Should I set probe like this if there's no probe info in the recording object?
[image]

samuelgarcia, 2021-11-19:
Yes, you need to attach a probebefore any run_sorter.

Yuhao-Zhang-SV, 2021-12-09:
So I'm currently using the recording with probe information. But I still get the same error as the recording without probe info. So what could that issue be? Thanks in advance~ We are trying to migrate from version **0.13** to the **newest** version of spikeinterface. (**SpikeSortingError: Spike sorting failed.**)
[image]

samuelgarcia, 2021-12-09:
What do you have when you do `rec.get_property('location')` ?
Normaly after `rec = rec.set_probe(a_probe)'` the location property is set automatically.

Yuhao-Zhang-SV, 2021-12-09:
OK, thanks! Actually it's None before, and then I managed to use a NumpyRecording object with available and correct get_property('location') info and it works! Thanks again.

closed 2021-12-11 by Yuhao-Zhang-SV