SpikeInterface/spikeinterface #382: Problem with run_kilosort() and read_kilosort().

szbalazs98, 2021-11-23:
I would like to retrieve ground-truth data by manual curation on sorted data by kilosort. When I run this code:
 sorting_KS = ss.run_kilosort(recording_saved, output_folder='results_KS_GT', 
                                verbose=True)
I get this error:
```
~\anaconda3\lib\site-packages\spikeinterface\extractors\phykilosortextractors.py` in __init__(self, folder_path, exclude_cluster_groups, keep_good_only)
     72                     cluster_info = pd.merge(cluster_info, new_property, on='cluster_id')
     73 
---> 74             cluster_info["id"] = cluster_info["cluster_id"]
     75             del cluster_info["cluster_id"]
     76

TypeError: 'NoneType' object is not subscriptable
```

When I try to read a previosly sorted and curated data with this code:
sorting_KS2_GT = se.read_kilosort(folder_path='./result_KS_GT',keep_good_only=True)

I get this error:
```
~\anaconda3\lib\site-packages\pandas\core\indexes\base.py in get_loc(self, key, method, tolerance)
   3080                 return self._engine.get_loc(casted_key)
   3081             except KeyError as err:
-> 3082                 raise KeyError(key) from err
   3083 
   3084         if tolerance is not None:

KeyError: 'cluster_id'
```

The error must something to do with 'cluster_id'.

Alternatively it would be much easier if I could read Kilosort2 manually curated data, but I couldn't find the function to import it. Am I right?

Thank you for your answer in advance!

szbalazs98, 2021-11-24:
I think I fixed it by using this command:
`sorting_KS_GT = se.PhySortingExtractor(folder_path='./results_KS2_ground_truth',exclude_cluster_groups="noise")`
But when I try to use `compare_sorter_to_ground_truth()`, I get this error:
`ValueError: all the input arrays must have same number of dimensions, but the array at index 0 has 2 dimension(s) and the array at index 1 has 1 dimension(s)`

alejoe91, 2021-11-24:
Hi @szbalazs98 

Can you share the code you used for comparison and the entire Traceback?

Cheers
Alessio

szbalazs98, 2021-11-24:
Of course.

I tried to run it three ways:
`c2gt = sc.compare_sorter_to_ground_truth(gt_sorting=sorting_KS_GT, tested_sorting=sorting_IC, verbose=True)`

```
mcmp = sc.compare_multiple_sorters([sorting_KS_GT, sorting_HS, sorting_IC, sorting_SC, sorting_KS2, sorting_TDS], ['Ground-truth','HS', 'IC', 'SC', 'KS2', 'TD'], 
                                   spiketrain_mode='union', verbose=True)
```
`study.run_comparisons(exhaustive_gt=True, match_score=0.1)
`
And I've got this Traceback:
```
Traceback (most recent call last):
  File "C:\Users\SortingStation\anaconda3\lib\site-packages\joblib\externals\loky\process_executor.py", line 431, in _process_worker
    r = call_item()
  File "C:\Users\SortingStation\anaconda3\lib\site-packages\joblib\externals\loky\process_executor.py", line 285, in __call__
    return self.fn(*self.args, **self.kwargs)
  File "C:\Users\SortingStation\anaconda3\lib\site-packages\joblib\_parallel_backends.py", line 595, in __call__
    return self.func(*args, **kwargs)
  File "C:\Users\SortingStation\anaconda3\lib\site-packages\joblib\parallel.py", line 262, in __call__
    return [func(*args, **kwargs)
  File "C:\Users\SortingStation\anaconda3\lib\site-packages\joblib\parallel.py", line 262, in <listcomp>
    return [func(*args, **kwargs)
  File "C:\Users\SortingStation\anaconda3\lib\site-packages\spikeinterface\comparison\comparisontools.py", line 103, in count_match_spikes
    num_matches = count_matching_events(times1, times2, delta=delta_frames)
  File "C:\Users\SortingStation\anaconda3\lib\site-packages\spikeinterface\comparison\comparisontools.py", line 29, in count_matching_events
    times_concat = np.concatenate((times1, times2))
  File "<__array_function__ internals>", line 5, in concatenate
ValueError: all the input arrays must have same number of dimensions, but the array at index 0 has 2 dimension(s) and the array at index 1 has 1 dimension(s)
"""

The above exception was the direct cause of the following exception:

ValueError                                Traceback (most recent call last)
<ipython-input-18-ae92333684be> in <module>
----> 1 c2gt = sc.compare_sorter_to_ground_truth(gt_sorting=sorting_KS_GT, tested_sorting=sorting_IC, verbose=True)

~\anaconda3\lib\site-packages\spikeinterface\comparison\groundtruthcomparison.py in compare_sorter_to_ground_truth(*args, **kwargs)
    466 
    467 def compare_sorter_to_ground_truth(*args, **kwargs):
--> 468     return GroundTruthComparison(*args, **kwargs)
    469 
    470 

~\anaconda3\lib\site-packages\spikeinterface\comparison\groundtruthcomparison.py in __init__(self, gt_sorting, tested_sorting, gt_name, tested_name, delta_time, sampling_frequency, match_score, well_detected_score, redundant_score, overmerged_score, chance_score, exhaustive_gt, n_jobs, match_mode, compute_labels, compute_misclassifications, verbose)
     79         if tested_name is None:
     80             tested_name = 'tested'
---> 81         BaseTwoSorterComparison.__init__(self, gt_sorting, tested_sorting, sorting1_name=gt_name,
     82                                          sorting2_name=tested_name, delta_time=delta_time,
     83                                          match_score=match_score,  # sampling_frequency=sampling_frequency,

~\anaconda3\lib\site-packages\spikeinterface\comparison\basecomparison.py in __init__(self, sorting1, sorting2, sorting1_name, sorting2_name, delta_time, match_score, chance_score, n_jobs, verbose)
     74         self.unit2_ids = self.sorting2.get_unit_ids()
     75 
---> 76         self._do_agreement()
     77         self._do_matching()
     78 

~\anaconda3\lib\site-packages\spikeinterface\comparison\basecomparison.py in _do_agreement(self)
    103 
    104         # matrix of  event match count for each pair
--> 105         self.match_event_count = make_match_count_matrix(self.sorting1, self.sorting2, self.delta_frames,
    106                                                          n_jobs=self._n_jobs)
    107 

~\anaconda3\lib\site-packages\spikeinterface\comparison\comparisontools.py in make_match_count_matrix(sorting1, sorting2, delta_frames, n_jobs)
    139     s2_spiketrains = [sorting2.get_unit_spike_train(u2) for u2 in unit2_ids]
    140 
--> 141     match_event_count_lists = Parallel(n_jobs=n_jobs)(delayed(count_match_spikes)(sorting1.get_unit_spike_train(u1),
[... 15 lines cut ...]
    935                     self._output.extend(job.get())

~\anaconda3\lib\site-packages\joblib\_parallel_backends.py in wrap_future_result(future, timeout)
    540         AsyncResults.get from multiprocessing."""
    541         try:
--> 542             return future.result(timeout=timeout)
    543         except CfTimeoutError as e:
    544             raise TimeoutError from e

~\anaconda3\lib\concurrent\futures\_base.py in result(self, timeout)
    437                 raise CancelledError()
    438             elif self._state == FINISHED:
--> 439                 return self.__get_result()
    440             else:
    441                 raise TimeoutError()

~\anaconda3\lib\concurrent\futures\_base.py in __get_result(self)
    386     def __get_result(self):
    387         if self._exception:
--> 388             raise self._exception
    389         else:
    390             return self._result

ValueError: all the input arrays must have same number of dimensions, but the array at index 0 has 2 dimension(s) and the array at index 1 has 1 dimension(s)
```

szbalazs98, 2021-12-01:
Dear @alejoe91 !

So there aren't any fix to this problem?
I don't want to be pushy, but it iss quite crucial to my work.
Thank you in advance!

alejoe91, 2021-12-01:
Can you try to set `n_jobs=1` when you call the `compare_sorter_to_ground_truth` function? Also, could you export all the sortings to npz and send them to me? 

You can do:
```
sorting_saved = sorting_KS.save(folder="sorting_KS")
```

This will create a folder with an `.npz` file inside. You can do this for all sorting objects, zip them, and send them so I can take a look.

szbalazs98, 2021-12-01:
[files.zip](https://github.com/SpikeInterface/spikeinterface/files/7635430/files.zip)
The same problem occured with n_jobs=1.

I attached the files. It contains the ground truth which is a Phy sorted Spyking Circus sorting, and the tested sorting which is generated by Kilosort2.

It seems to me that PhySortingExtractor() causes this problem, because it occurs when we use Phy on Kilosort2, Kilosort1 and Spyking Circus sortings.

alejoe91, 2021-12-01:
Ok the problem seems to be that the Phy object was not loaded properly. The spike trains show 165882 spikes for unit 0 and 0 spikes for all others...Can you share the phy folder on drive so I can investigate it further? Also, the same problem arise from the KS sorting. This is likely why the comparison is failing...

szbalazs98, 2021-12-01:
Of course: Phy folder: https://drive.google.com/drive/folders/13TMT0-K3soeJPFGz1KbBYTKXUZ2nSE-y?usp=sharing
When I use `sorting_loaded_KS.get_unit_spike_train(unit_id=3)`, it seems fine to me with KS.
Can this be a problem: before I use PhySortingExtractor I change the first line (from cluster_id to id) in cluster_info.txt, because if I don't an other error occurs.

closed 2021-12-02 by samuelgarcia