SpikeInterface/spikeinterface #384: Data loading of Intan data

MNS57, 2021-11-25:
Dear Buccino

Now, we are going to load intan data through the function read_intan.
Then, how should we write the referring "stream_id"?

Data names are like A_154010.rhd, A_154110.rhd, A_154210.rhd,,,, A_169011.rhd, A_161011.rhd

Thanks.

Nori,

alejoe91, 2021-11-25:
Hi Nori,

Thank you for the issue. Do all the files refer to the same recording?

As for the stream, I think you should use `stream_id='0'`. Right @samuelgarcia ?

Cheers
Alessio

MNS57, 2021-11-25:
We still meet errors, do you find any remaining problems?

-------------------------------------------------------------------------------------------------------
[input]
local_path = '/home/MSNNlab/SpikeData/Default_InViVo/211019/100/data02_211019_154010.rhd'
recording, sorting_true = se.read_intan(local_path, stream_id='0')

-------------------------------------------------------------------------------------------------------
[feedback]

['0' '1' '2']
0
Traceback (most recent call last):

  File "<ipython-input-53-8d0e642e91c8>", line 3, in <module>
    recording, sorting_true = se.read_intan(local_path, stream_id='0')

TypeError: cannot unpack non-iterable IntanRecordingExtractor object

MNS57, 2021-11-25:
> Do all the files refer to the same recording?

The answer for this question is yes.

alejoe91, 2021-11-25:
Hi, the problem is that Intan only returns a `recording` object, not a sorting one as well.

So you should do: 
```
recording = se.read_intan(local_path, stream_id='0')
```

alejoe91, 2021-11-25:
> > Do all the files refer to the same recording?
> 
> The answer for this question is yes.

Ok , in that case you need to sort all files and have them in a list. Then you can load them sequentially and concatenate the recordings:

```
files_list = ['file1...', 'file2...', ...]

recording_list = []
for file in files_list:
    recording = se.read_intan(file)
    recording_list.append(recording)

recording_concat = si.concatenate_recordings(recording_list)
```

MNS57, 2021-11-25:
Great! We wanted to ask it too.

Now, we could run up to the before half of "Getting started tutorial".

Let us ask something later.

Thanks!

MNS57, 2021-12-08:
Hi,

Could we ask two more questions?

1. Could you kindly partly write how can we specifically set channel positions using " .set_device_channel_indices"?
2. How can we load a reference signal as well as "se.read_intan(file, stream_id='0')"?

Thanks.

alejoe91, 2021-12-19:
Hi @MNS57 

> 1. Could you kindly partly write how can we specifically set channel positions using " .set_device_channel_indices"?

The `device_channel_indices` are the mapping between the electrodes and the recorded signals. Do you have a channel map for your data? What probe and connector are you using?

> 2. How can we load a reference signal as well as "se.read_intan(file, stream_id='0')"?

What kind of reference signal? How do you record it? If you want to re-reference your data, you can use the `st.common_reference` function. It enables to use either a common reference (e.g. median) or to rereference using a single channel

MNS57, 2021-12-20:
Hi @alejoe91

1. Although this is a solution from a different aspect of your answer, if the map shown in this code 
    is same as our known spatial map of recording points' indecs, does it mean we gave the correct 
    definition of recording points' indecs?

plot_probe(probe, with_device_index=True, ax = ax)

2. So, if we used a single external reference channel, is this the right code?

recording_single = st.common_reference(recording, reference='single', ref_channels=[1])

Thanks.

alejoe91, 2021-12-20:
> Hi @alejoe91
>
> 1. Although this is a solution from a different aspect of your answer, if the map shown in this code
>    is same as our known spatial map of recording points' indecs, does it mean we [...]

Yes, if thie device indices correspond to what you expect from your configuration than it's correct.

> 2. So, if we used a single external reference channel, is this the right code?
> 
> recording_single = st.common_reference(recording, reference='single', ref_channels=[1])

Yes, this is the right code tp rereference everithing to channel `1`. However, I would also suggest you can try to use common median reference as it usually removes a lot of common noise:

recording_cmr = st.common_reference(recording, reference='global', operator='median')

Cheers
Alessio

MNS57, 2021-12-25:
Thank. We could start a spike sorting!

Then, in running a spike sorter like:
sorting_SS = ss.run_spykingcircus(recording=recording_f_clean, detect_threshold=4)

We met this following error... how we can solve it? 
    File ".../extractors/spykingcircusextractors.py", line 53, in __init__
    raise Exception(spykingcircus_folder, " is not a spyking circus folder")
    Exception: (PosixPath('spykingcircus_output/recording'), ' is not a spyking circus folder')

In addition, how can we change the necessary parameters from the default values following the code?
  ss.get_default_params('spykingcircus')

Thanks.

alejoe91, 2021-12-25:
Hi, 

Seems to be a spiking circus problem. Please refer to the spyking circus documentation and open an issue there.

For changing parameters, you can just pass additional parameters in the `run_` function (see here for an example: https://spikeinterface.readthedocs.io/en/latest/getting_started/plot_getting_started.html )

MNS57, 2021-12-26:
Hi,   

Thanks.  

About the second issue, we will be able to set parameters. 
But, we could not change the parameter called "max_clusters"... is the name slightly different in spikeintereface?

About the first issue, I suppose this file "..../spikeinterface/extractors/spykingcircusextractors.py" is one file developed in spikeinterface... is how we can interpret it as an issue in spiking circus?

alejoe91, 2021-12-26:
You can use the `ss.get_params_description('spkyngcircus)` to get a list of the available parameters and description for a sorter.

The problem is that spykingcircus is non finishing correctly, so Spikeinterface cannot load the output. You can check the `log` file in the output folder to see the output from spykingcircus and you'll see some errors are raised there, so spyking circus is failing. Maybe it's not being installed properly? You can ask on the spyking circus GitHub page by attaching the log.

MNS57, 2021-12-28:
Sure. I ask the developer of spyking circus ... Sorry, I'm not familiar with the usage of GitHub, so we will ask by e-mail.

At your side, could you add the option of "max_clusters" in the program of ss.run_spykingcircus?   It's an essential parameter in our understanding.

alejoe91, 2021-12-28:
> Sure. I ask the developer of spyking circus ... Sorry, I'm not familiar with the usage of GitHub, so we will ask by e-mail.
> 

You can open an issue from here: https://github.com/spyking-circus/spyking-circus/issues

> At your side, could you add the option of "max_clusters" in the program of ss.run_spykingcircus? It's an essential parameter in our understanding.

I don't find the `max_clusters` parameter. See SpykingCircus config file here: https://spyking-circus.readthedocs.io/en/latest/code/config.html
Can you point me to it?

MNS57, 2021-12-28:
> You can open an issue from here: https://github.com/spyking-circus/spyking-circus/issues

OK. Although we once asked directly by e-mail, we will add the following issue there.

> I don't find the max_clusters parameter.

This may be one of reference pages about the parameter, although I don't know if this is the best page:
https://spyking-circus.readthedocs.io/en/latest/advanced/tuning.html

alejoe91, 2021-12-28:
> > I don't find the max_clusters parameter.
> 
> This may be one of reference pages about the parameter, although I don't know if this is the best page: https://spyking-circus.readthedocs.io/en/latest/advanced/tuning.html

Thanks. Still I can't find it in the [params file](https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/sorters/spyking_circus/config_default.params) that we ported from spyking circus, so maybe this is another thing to ask on the spyking circus page.

MNS57, 2021-12-28:
>  Still I can't find it in the params file that we ported from spyking circus,

I think you need to add "max_clusters   =  ~~" in the section of [clustering]
in the parameter file as well as other parameters.  

Does spikeinterface produce such parameter file internally?

alejoe91, 2021-12-28:
It's copied from the spykingcircus repo: https://github.com/spyking-circus/spyking-circus/blob/master/circus/config.params

alejoe91, 2021-12-28:
@yger what is the default to `max_clusters`? Is there a reason why it's not in the config.params in the SC repo?

closed 2022-01-07 by alejoe91