SpikeInterface/spikeinterface #385: Compute ISI violation metric defined in Hill et al. (2011)

Yizhar-Lab, 2021-11-25:
The Spike interface (SI) [quality metrics tutorial](https://spikeinterface.readthedocs.io/en/latest/module_toolkit.html#quality-metrics) references [these excellent docs from the Allen Institute](https://allensdk.readthedocs.io/en/latest/_static/examples/nb/ecephys_quality_metrics.html), which in turn references the metrics defined in [Hill et al. (2011) J Neurosci 31: 8699-8705](https://www.jneurosci.org/content/31/24/8699). However, I've only managed to find SI functions that compute the total number of ISI violations or the rate (the count divided by the recording duration). Is there a way to compute this alternative metric?

alejoe91, 2021-11-25:
Hi @Yizhar-Lab 

You are right! Sorry we have refactored the code lately and I think @samuelgarcia did not implement the ratio described in Hill 2011. Thanks for spotting it, we'll update the code so that the ratio actually refers to the Hill metric.

Cheers
Alessio

Yizhar-Lab, 2021-11-25:
That's great! And thanks for the super fast reply. 
Is there a way to monitor for this update, or should I just check the code every now and then?

alejoe91, 2021-11-25:
I'm working on it now, will make a PR in a few minutes :) You should get an email everytime I comment or reference this issue (I'll point to the issue in the PR)

Dradeliomecus, 2021-11-25:
Actually the formula described in Hill et Al. 2011 underestimates the contamination (and the underestimation becomes greater as the contamination rises).
They actually end up with noise coming from a single neuron rather than random noise.

For the case of random noise, the estimated contamination should be:
C = 1 - sqrt(1 - n_v * T / (t_r * N^2))
Where C is the contamination, n_v the number of refractory violations, T the length of the recording, t_r the refractory period (same unit as T) and N the total number of spikes in the cluster

In the case of a single unit bringing the contamination to the cluster, then the formula becomes
C = (1 - sqrt(1 - 2*n_v*T / (t_r * N^2))) / 2

These formulas are included in our preprint, which we should submit soon.
Hope it helps

samuelgarcia, 2021-11-25:
@alejoe91 : I don't think you implement this.
I propose having several metrics for this idea : the simple counting, the simple rate (as before), the hill2011 and this one.
The NamedTuple allow to return several metrics for one computation.

samuelgarcia, 2021-11-26:
@Dradeliomecus  : we could implement this but how could be the name ?

Dradeliomecus, 2021-11-26:
@samuelgarcia I guess something like `estimated_contamination` ?
If you're talking about a reference, the preprint should be sent in december if everything goes right, if not early 2022.

samuelgarcia, 2021-11-26:
ok thanks.
I propose to finish this PR #386 no and add your metric as soon as the preprint is out.

closed 2021-12-25 by alejoe91