SpikeInterface/spikeinterface #387: principle component computation concern

rtraghavan, 2021-11-28:
In one of my recent recordings the compute_principle_components module failed to run due to the following error

n_components=3 must be less or equal to the batch number of samples 1

Tracking down the issue it turned to be an error in principle_component.py in the spike interface toolkits post processing folder and was getting caught on one of the units which extract_waveforms yieled a single waveform or a  it was a 1 x time x channels matrix. The code has the following line in it to catch empty channels

if wfs.size==0:
       continue

But for mostly empty channels do you forsee any issue with modifying this as follows

if np.size(wfs,0) <= p['n_components']:
       continue

I think this would solve this issue which mostly has to do with the nitty gritty of the incremental PCA algorithm. 

Alternatively can you suggest a different approach. Perhaps the global flag in compute_principle_components could be used, but I'm not sure that is helpful in this case.

alejoe91, 2021-11-29:
Hi @rtraghavan 

Thanks for spotting this. Your solution would prevent the code from failing, but since we fit a different PCA model for each unit I think that it would be best to compute the model only if there is a minimum number of waveforms for the unit (say 50 by default). If you had 5 spikes/waveforms and compute a PCA model with 5 components the PCA would't fail, but I wouldn't *trust* the PCA model. What do you think?

rtraghavan, 2021-11-29:
This seems reasonable, though perhaps it is better to choose the minimum number of waveforms based on the number of components. Something like 10-20 x n_components is roughly in line with what you recommend and seems to be what people suggest for factor analysis and related analyses online. Though I know of no hard or fast rules and perhaps it is best to start with some default (say scale = 10) and leave it up to the person using spikeinterface in order to decide?

alejoe91, 2021-11-30:
Sounds good! Do you feel like making a pull request? :)

rtraghavan, 2021-12-17:
Working on it now. Will hopefully come through soon.

alejoe91, 2022-01-13:
https://github.com/SpikeInterface/spikeinterface/issues/418

guitchounts, 2022-01-24:
Hi everyone, was this issue solved? I'm running into the same problem and this is affecting downstream computation of quality metrics. While each of my extracted units has multiple spike times (e.g. lowest is a unit with 11 spikes, but many have hundreds or thousands of spikes), for whatever reason the saved waveforms have 0 spikes for a bunch of units. Is the PC error due to there being no spikes in from the extracted waveforms for some units?

alejoe91, 2022-04-01:
https://github.com/SpikeInterface/spikeinterface/pull/571

closed 2022-04-01 by alejoe91