SpikeInterface/spikeinterface #393: recording.set_channel_locations fails

Yizhar-Lab, 2021-12-02:
I just upgraded to the newest version and tried to run the following code, which used to run with no problem before the upgrade.
`recording.set_channel_locations(positions, channel_ids=list(range(n)))`
where 'positions' is just:
array([[   0,    0],
       [   0,  150],
       [   0,  300],
       [   0,  450],
       [   0,  600],
       [   0,  750],
       [   0,  900],
       [   0, 1050],
       [   0, 1200],
       [   0, 1350],
       [   0, 1500],
       [   0, 1650],
       [   0, 1800],
       [   0, 1950],
       [   0, 2100],
       [   0, 2250]]) 
but now I get the error: 
ValueError: 0 is not in list
The full error message is in the following text file.
[rec.set_channel_locations error.txt](https://github.com/SpikeInterface/spikeinterface/files/7643533/rec.set_channel_locations.error.txt)
Thanks!

alejoe91, 2021-12-02:
Hi @Yizhar-Lab 

Can you try this: `recording.set_channel_locations(positions)` if the locations are defined for all channels? 

Also, what's the output of `recording.get_channel_ids()`? I suspect that the channel ids are not `range(n)`

Yizhar-Lab, 2021-12-06:
Hi,
`recording.get_channel_ids()` returns
```
array(['0', '1', '10', '11', '12', '13', '14', '15', '2', '3', '4', '5',
       '6', '7', '8', '9'], dtype='<U64')
```
`recording.set_channel_locations(positions)` seems to work.
Thanks!

closed 2022-01-07 by alejoe91