SpikeInterface/spikeinterface #402: Control KS2 outputs from SpikeInterface?

Yizhar-Lab, 2021-12-07:
Hi,
Is it possible to do the following using SpikeInterface parameters (sorry if it's basic, my quick search in the docs didn't yield anything)?
- Stop KS2 from copying the recorded neural data to `recording.dat` and `temp_wh.dat`, or removing them at the end of sorting. Of course I can delete them myself, but that'd require changing other files and features to tell downstream analysis (e.g. phy) to look for the raw data files in their original location. Can SpikeInterface handle this inherently? 
- Add unit qulity metrics to the files used by phy so that phy displays these metrics in the cluster (or other) view
- Remove units based on some criteria so that they are not shown in phy

Thanks!

Yizhar-Lab, 2021-12-07:
Actually, the [kilosort forums say](https://www.giters.com/MouseLand/Kilosort/issues/287) one shouldn't remove these "copies" of the raw data because they're drift corrected.

alejoe91, 2021-12-13:
> Hi, Is it possible to do the following using SpikeInterface parameters (sorry if it's basic, my quick search in the docs didn't yield anything)?
>
> * Stop KS2 from copying the recorded neural data to [...]

I guess that given your comment it's better to keep them then!

> * Add unit qulity metrics to the files used by phy so that phy displays these metrics in the cluster (or other) view

This indeed could be done. We would first need to load the quality metrics as properties. Could you open a separate issue about that?

> * Remove units based on some criteria so that they are not shown in phy

After computing metrics, you can query the dataframe with any rule to only select units that you want to keep. Then you can use `select_units()` to retrieve only the units that you are interested in and `export_to_phy` using the new `sorting` (returned by `select_units`). (see [this tutorial](https://github.com/SpikeInterface/spiketutorials/blob/master/Official_Tutorial_SI_0.90_Aug2021/SpikeInterface_Tutorial.ipynb) - Validation and Curation - seciton 7)

> Thanks!

alejoe91, 2022-01-13:
@Yizhar-Lab 

- Phy exports quality metrics with this PR: https://github.com/SpikeInterface/spikeinterface/pull/450
- The WaveformExtractor `select_units` has been implemented (https://github.com/SpikeInterface/spikeinterface/pull/443 and https://github.com/SpikeInterface/spikeinterface/pull/448)

Closing this

closed 2022-01-13 by alejoe91