SpikeInterface/spikeinterface #403: How to inspect ProbeGroup object's field info & How to turn ProbeGroup into Probe?

Yuhao-Zhang-SV, 2021-12-08:
After reading .prb file into a ProbeGroup object, how do I inspect the content of the object (in Google Colab debug mode using ipdb)? Where can I find the definition of the data structure of **ProbeGroup** class? Since I can only see the class methods in API document.

Besides, I want to _set_probe(probe)_ for a recording, but it needs _probe_ to be Probe object instead of ProbeGroup object, so what could I do for the transition from **ProbeGroup** to **Probe** object? Thanks in advance~

samuelgarcia, 2021-12-08:
`ProbeGroup` is basocally a list of `Probe` you can acces with `probegroup.probes`.
So with a loop you can inspect the probe itself
```
for probe in probegroup.probes:
   print(probe)
   print(probe.to_dataframe())
```

You also have a `recording.set_probegroup(group_mode='by_probe' or 'by_shank')`

We have lots of example to understand the probeinterface https://probeinterface.readthedocs.io/en/main/examples/index.html

A very important aspcet is the wiring : https://probeinterface.readthedocs.io/en/main/examples/ex_11_automatic_wiring.html#sphx-glr-examples-ex-11-automatic-wiring-py

Note that using your old PRB file is not really a way to handle probe and channel map.
Because the "group" concept in PRB is very very ambiguous about if a group is a probe or a shank.
Maybe you should handle your probe directly without your old PRB file. You would have a better control on the geometry and channel mapping.

Are you using silicon probes ? If yes then use our probe library. 
Do you have tetrode ? If yes also handle directly your probe in with `Probe` like in examples.

Hope it helps.

Yuhao-Zhang-SV, 2021-12-09:
Well thanks it's of great help to me!! BTW, what's the differences between '**by_probe**' and '**by_shank**' for _set_probegroup_?

samuelgarcia, 2021-12-09:
It handle the way the grouping is done.
If you have 2 probes with 4 shanks each : with 'by_probe' you will get 2 groups and with 'by_shank' you will get 8 groups.
Then when you split (with `recordings_by_group = recording.split_by("group")`) you can make independant sorting on ether probe or shank.

Have a look to
https://github.com/SpikeInterface/spiketutorials/blob/master/Official_Tutorial_SI_0.90_Aug2021/SpikeInterface_Tutorial.ipynb

closed 2021-12-25 by alejoe91