SpikeInterface/spikeinterface #404: Kilosort2Sorter.get_result_from_folder raise error

AntonioST, 2021-12-09:
Hi,

I used `ss.Kilosort2Sorter.get_result_from_folder` to load my kilosort result from disk, and it raised a KeyError, because "error" key is not in the spikeinterface_log.json.

```
Traceback (most recent call last):
  ...
  File ".../util_compare.py", line 25, in cmp_result
    r1 = ss.Kilosort2Sorter.get_result_from_folder(k1.directory)
  File ".../spikeinterface/sorters/basesorter.py", line 252, in get_result_from_folder
    if bool(log['error']):
KeyError: 'error'
```

[spikeinterface_log.txt](https://github.com/SpikeInterface/spikeinterface/files/7685914/spikeinterface_log.txt)

AntonioST, 2021-12-09:
And also I find it is hard to send `keep_good_only` into `BasePhyKilosortSortingExtractor`, so I need to prepare another `spikeinterface_params.json` with `keep_good_only` set to true. Is it able to add keyword parameters on `get_result_from_folder`? 

And the good cluster query expression at line `BasePhyKilosortSortingExtractor:89`. Does it should be `"KSLabel == 'good"`?

```
if keep_good_only and "KSLabel" in cluster_info.columns:
    cluster_info = cluster_info.query("KSLabel != 'good'")
```
The class named with Phy, so I think it should get good cluster information from phy's result (from column "group" in `cluster_info.tsv`). How do I change the "KSLabel" to "group" for `keep_good_only`?

AntonioST, 2021-12-09:
The shape of `spike_times.npy` is `(N, 1)`, and it should be flatten to `(N,)`. Otherwise, `get_unit_spike_train` will give me a wrong shape array and I will get an all zero matrix result from `compare_two_sorters`. Even if I used same source result.

samuelgarcia, 2021-12-13:
@alejoe91 : I let you check this ?

alejoe91, 2021-12-13:
Hi @AntonioST 

Did you run KS from spikeinterface, or externally? In case you ran it with SI, then the `log['error']` should be there (https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/sorters/basesorter.py#L199-L211).
If you ran it externally, you can use the `KiloSortSortingExtractor` directly.

You can use the `keep_good_only=True` directly as an extra argument when you call KS. This will be saved in the sorting params and used by the `get_result_from_folder()` function (see https://github.com/SpikeInterface/spikeinterface/blob/6c9c7813c3b90c1e1c23b52535726f89f43e5dae/spikeinterface/sorters/kilosortbase.py#L45-L51)

Thanks for spotting the wrong query!!! Made a fix here: 
https://github.com/SpikeInterface/spikeinterface/pull/409

The dimension error was already fixed from this PR:
https://github.com/SpikeInterface/spikeinterface/pull/392

closed 2021-12-13 by samuelgarcia