SpikeInterface/spikeinterface #407: The provenace is not dumpable

Yuhao-Zhang-SV, 2021-12-11:
When I `save `a `recording_cmr` object, it generated a file named `provenance.json` and the file only contains a warning `{"warning": "the provenace is not dumpable!!!"}`. But other files seems normal. What could be the problems? I've set probe info. Thanks in advance~

samuelgarcia, 2021-12-13:
This is a warning. It informs you that you the source of recording cannot be describe with the json format because certainly at some point it has a NumpyReccording

Yuhao-Zhang-SV, 2021-12-14:
So basically NumpyRecording is not dumpable right? So why is it not dumpable? And which kind of recording object is dumpable? Thanks~

alejoe91, 2021-12-14:
@Yuhao-Zhang-SV dumpable.means that the description of the object can be saved to dict or a json file and distributed across jobs. All objects loaded from a file are dump able, and this allows us to reload them in different jobs/processes from the same files. Numpy recordings/sorting are not dump able because you would need to add the entire data to the dict/json file, which is highly inefficient.

samuelgarcia, 2021-12-14:
In short NumpyRecording is the only none-dumpable source.
But some processing steps can also make the recording none-dumpable even most of then are dumpable.

closed 2021-12-25 by alejoe91