SpikeInterface/spikeinterface #410: Recording filtering and appending using ProbeGroup

emdezla, 2021-12-14:
The probe I am using combines both **tetrodes** and **single electrodes** so I create a `ProbeGroup` for separating different channel groups. However, once I separate the recording by channel group as in this [tutorial](https://spikeinterface.readthedocs.io/en/latest/modules/sorters/plot_3_sorting_by_group.html), I am not able to filter the data and apply common-mode rejection because `st.bandpass_filter` or `st.common_reference` does not accept `recording_by_group` as an argument.

I tried to run a loop and filter one recording at each iteration but then I end up with a list of recording segments that is not supported by the method `run_sorters`. Is there a method for treating this special case or even for appending recordings with a different number of channels?

samuelgarcia, 2021-12-14:
Normally, SI is flexible to have make several use case:
 1. `split>filter>cmr` : here you need to loop.
 2. `filter>cmr>split` : here no loop because the filtering+cmr is done on on object

`run_sorters` accept both a dict or list of recording.

Normally `split` do not give list of segment but dict/list of sub recording (split on channels).

Could you show some code ?

alejoe91, 2021-12-14:
Hi Emilio,

Your approach is correct, you can first split_by group, then apply preprocessing steps (the filtering step can be done before), and finally aggregate the output recording. Note that you can run spike sorting by property with a specific function. Here is the code:
```
# recording with group property is called recording_full
recording_full_filt = st.bandpass_filter(recording_full)
recording_split_filt = recording_full_filt.split_by("group")

# loop and apply cmr
recordings_list = []
for group, recording_split in recording_split_filt.items():
    recording_split_cmr = st.common_reference(recording_split)
    recording_list.append(recording_split_cmr)

# re-aggregate the recording
recording_agg = si.aggregate_channels(recording_list)
# make sure groups are still there
print(recording_agg.get_channel_groups())

# spike sort by property
sorting_group = ss.run_sorter_by_property("kilosort2", recording_agg, grouping_property="group", working_folder="my_output_by_group")
```

The `aggregate_channels` allows you to concatenate channels over channels, which is what you need. YOu might need to upgrade spikeinterface to use it (`pip install --upgrade spikeinterface`)

emdezla, 2021-12-15:
Thank you very much for your answers @samuelgarcia @alejoe91 

I am still getting errors that most likely arise from my probe architecture, so let me give more details about it. My probe is a 32-channel probe (4 tetrodes and 16 single electrodes). In the previous version of SpikeInterface, I used to create a group for every tetrode and every single electrode, resulting in 20 channel groups in total that were sorted independently but then could be processed and exported to Phy together. 

However, with the recent updates of SpikeInterface and especially the addition of the ProbeInterface package, I am no longer sure this is the best approach given that I get several errors or inconveniences related to the fact that some groups are only comprised of one channel. 

For example, after running  `st.common_reference(recording_split)`, every single electrode channel just vanishes. My guess is that the channel is being removed because it is being identified as the reference channel and I tried to apply CMR with different parameters such as `reference=local` but it is still not working. 

Anyways, even if I run the spike sorter without any prior processing (just as the raw recording split by groups), I still have a hard time trying to visualise and export the data to other platforms such as Phy for manual curation.  Running `sorting_group = ss.run_sorter_by_property("ironclust", recording_agg, grouping_property="group", 
                                            working_folder="my_output_by_group", mode_if_folder_exists='overwrite')` outputs a dict whose items are in the form of `{('0', 'ironclust'): MdaSortingExtractor: 1 units - 1 segments - 30.0kHz`. This is just fine if I just want to play with one group at a time, but I would like to play with all the groups at the same time and I can't because the attributes are not adapted. 
```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
~\AppData\Local\Temp/ipykernel_10960/1576065133.py in <module>
----> 1 print(f'Ironclust unit ids: {sorting_IC.get_unit_ids()}')

AttributeError: 'dict' object has no attribute 'get_unit_ids'
``` 
So my general question is: are there indivdiual solutions for each of my problems in order to adapt the code to having multiple channel groups or am I missing the bigger picture here and there is a better approach for defining my channels and probe architecure at the beggining of the code?

alejoe91, 2022-01-07:
Hi @emdezla 

Note that you cannot use common reference on a dictionary (`st.common_reference(recording_split)`), but you need to loop over the recordings and apply CMR separately. Also note that you can;t apply CMR to a single channel because in that case the channel will become blank (the median of a single channel is the value of the trace).

Regarding the `run_sorter_by_property`, it doesn't return a dictionary, but a single `sorting` object (see [here](https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/sorters/launcher.py#L47-L136)). So maybe something else is wrong there?

closed 2022-04-01 by alejoe91