SpikeInterface/spikeinterface #418: suggestions for better memory management compute_pc for metrics

rtraghavan, 2021-12-17:
It isn't clear to me where the major bottleneck is in computing principle components, but it seems to be quite memory intensive as a process. For a 30 minute neurpixel array 3A recording, executing compute_principle_components has a memory utilization roughly 1.6-2X the file size (70-80 TB).

One way around this I have thought is to extract waveforms and principle components for clumps of channels and compute metrics for each clump, saving the metrics as I go. Perhaps in a future release the ability to compute PCs and metrics in chunks (of units) would be helpful. At least for the by_channel_local default option.

alejoe91, 2022-01-07:
HI @rtraghavan 

Sorry for our late reply and thanks for the issue! I think you mean 70-80 GB (not TB) right? 

One thing we are discussing is to add the possibility to compute PC on sparse channels (e.g. based on radius around max channel, or best channels - see [this function](https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/toolkit/postprocessing/template_tools.py#L93)). We'll keep you updated!

alejoe91, 2022-08-01:
Added parallelization and improved implementationin general

closed 2022-08-01 by alejoe91