SpikeInterface/spikeinterface #419: tridesclous errors on spikeinterface tutorial dataset

jjtukker, 2021-12-18:
hi,
I'm trying to work through the tutorial, on an Ubuntu server (xfce 4.14) and running into issues with several sorters (so far only mountainsort seems to work). For Trisdesclous, I get the following output:

sorting_TDC = ss.run_tridesclous(recording_saved, 
                              output_folder='results_TDC',
                              verbose=True)
print(f'Tridesclous found {len(sorting_TDC.get_unit_ids())} units')

DataIO <id: 140466321816976> 
  workdir: results_TDC
  sample_rate: 20000.0
  total_channel: 16
  channel_groups: 0 [ch0 ch1 ch2 ch3 ... ch12 ch13 ch14 ch15]
  nb_segment: 1
  length: 6000000
  durations: 300.0 s.
WARNING : peakdetector will be slow install opencl or numba
catalogue_nested_params
{'chunksize': 20000,
 'clean_cluster': {'apply_auto_merge_cluster': True,
                   'apply_auto_split': True,
                   'apply_trash_low_extremum': True,
                   'apply_trash_not_aligned': True,
                   'apply_trash_small_cluster': True},
 'clean_peaks': {'alien_value_threshold': -1.0, 'mode': 'extremum_amplitude'},
 'cluster_kargs': {'adjacency_radius_um': 50.0,
                   'high_adjacency_radius_um': 30.0,
                   'max_loop': 1000,
                   'min_cluster_size': 20},
 'cluster_method': 'pruningshears',
 'duration': 300.0,
 'extract_waveforms': {'wf_left_long_ms': -2.5,
                       'wf_left_ms': -1.0,
                       'wf_right_long_ms': 3.5,
                       'wf_right_ms': 1.5},
 'feature_kargs': {'adjacency_radius_um': 50.0, 'n_components_by_channel': 3},
 'feature_method': 'pca_by_channel',
 'make_catalogue': {'inter_sample_oversampling': False,
                    'sparse_thresh_level2': 3,
                    'subsample_ratio': 'auto'},
 'memory_mode': 'memmap',
 'mode': 'sparse',
 'n_jobs': -1,
 'n_spike_for_centroid': 350,
 'noise_snippet': {'nb_snippet': 300},
 'peak_detector': {'adjacency_radius_um': 200.0,
                   'engine': 'numpy',
                   'method': 'geometrical',
                   'peak_sign': '-',
                   'peak_span_ms': 0.7,
                   'relative_threshold': 5,
                   'smooth_radius_um': None},
 'peak_sampler': {'mode': 'rand_by_channel',
                  'nb_max': 20000,
                  'nb_max_by_channel': 1000},
 'preprocessor': {'common_ref_removal': False,
                  'engine': 'numpy',
                  'highpass_freq': 400.0,
                  'lowpass_freq': 5000.0,
                  'pad_width': -1,
                  'smooth_size': 0},
 'sparse_threshold': 1.5}
peeler_params
{'chunksize': 20000, 'engine': 'geometrical'}
estimate_signals_noise 0.7310680919326842
run_signalprocessor 130.1215224340558
clean_peaks 0.004330028779804707
sample_some_peaks 0.05209812195971608
extract_some_noise 0.13437372166663408
extract_some_features 1.8851443682797253
[... 280 lines cut ...]

~/miniconda3/envs/si/lib/python3.8/site-packages/spikeinterface-0.92.1.dev0-py3.8.egg/spikeinterface/sorters/runsorter.py in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, with_output, **sorter_params)
     55 
     56     if docker_image is None:
---> 57         sorting = run_sorter_local(sorter_name, recording, output_folder=output_folder,
     58                                    remove_existing_folder=remove_existing_folder,
     59                                    delete_output_folder=delete_output_folder,

~/miniconda3/envs/si/lib/python3.8/site-packages/spikeinterface-0.92.1.dev0-py3.8.egg/spikeinterface/sorters/runsorter.py in run_sorter_local(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, **sorter_params)
     84         recording, output_folder, sorter_params, verbose)
     85     SorterClass.setup_recording(recording, output_folder, verbose=verbose)
---> 86     SorterClass.run_from_folder(output_folder, raise_error, verbose)
     87     if with_output:
     88         sorting = SorterClass.get_result_from_folder(output_folder)

~/miniconda3/envs/si/lib/python3.8/site-packages/spikeinterface-0.92.1.dev0-py3.8.egg/spikeinterface/sorters/basesorter.py in run_from_folder(cls, output_folder, raise_error, verbose)
    234         if has_error and raise_error:
    235             print(log['error_trace'])
--> 236             raise SpikeSortingError(
    237                 f"Spike sorting failed. You can inspect the runtime trace in {output_folder}/spikeinterface_log.json")
    238 

SpikeSortingError: Spike sorting failed. You can inspect the runtime trace in results_TDC/spikeinterface_log.json

Unfortunately I have no idea what might be wrong, any ideas on how to proceed would be greatly appreciated! Thanks in advance.

alejoe91, 2021-12-19:
Hi, for TDC you need to install numba

`pip install numba`

samuelgarcia, 2022-01-03:
And also you will need maybe to downgrade numpy to 1.20

alejoe91, 2022-01-07:
does it help?

jjtukker, 2022-01-07:
YES it works now with numba (0.54.1) and numpy 1.20.3
Thanks a lot!

closed 2022-01-07 by jjtukker