SpikeInterface/spikeinterface #420: Kilosort3 errors on tutorial dataset

jjtukker, 2021-12-18:
hi,
I'm sorry for all the issues, but I just keep hitting a wall trying out different sorters from within the Jupyter notebook tutorial (Python 3.8), on Ubuntu server (xfce4.14), with the standard dataset provided. So far only Mountainsort4 works..

For Kilosort3, here is the command I run:
sorting_KS3 = ss.run_kilosort3(recording_saved, 
                              output_folder='results_KS3',
                              verbose=True)

Here is the output:
write_binary_recording: 100%|#####################| 1/1 [00:07<00:00,  7.13s/it]

KilosortBase._run_from_folder <class 'spikeinterface.sorters.kilosort3.kilosort3.Kilosort3Sorter'>
RUNNING SHELL SCRIPT: results_KS3/run_kilosort3.sh

                            < M A T L A B (R) >

                  Copyright 1984-2021 The MathWorks, Inc.

             R2021a Update 1 (9.10.0.1649659) 64-bit (glnxa64)

                               April 13, 2021

                            < M A T L A B (R) >

                  Copyright 1984-2021 The MathWorks, Inc.

             R2021a Update 1 (9.10.0.1649659) 64-bit (glnxa64)

                               April 13, 2021

 

 

To get started, type doc.

To get started, type doc.

For product information, visit www.mathworks.com.

For product information, visit www.mathworks.com.

 

 

    ----------------------------------------------------

	Your license will expire in 12 days.

	Please contact your system administrator or

	MathWorks to renew this license.

    ----------------------------------------------------

    ----------------------------------------------------

	Your license will expire in 12 days.

	Please contact your system administrator or

	MathWorks to renew this license.

    ----------------------------------------------------

welcome!

welcome!

[... 313 lines cut ...]
    371 
    372 

~/miniconda3/envs/si/lib/python3.8/site-packages/spikeinterface-0.92.1.dev0-py3.8.egg/spikeinterface/sorters/runsorter.py in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, with_output, **sorter_params)
     55 
     56     if docker_image is None:
---> 57         sorting = run_sorter_local(sorter_name, recording, output_folder=output_folder,
     58                                    remove_existing_folder=remove_existing_folder,
     59                                    delete_output_folder=delete_output_folder,

~/miniconda3/envs/si/lib/python3.8/site-packages/spikeinterface-0.92.1.dev0-py3.8.egg/spikeinterface/sorters/runsorter.py in run_sorter_local(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, **sorter_params)
     84         recording, output_folder, sorter_params, verbose)
     85     SorterClass.setup_recording(recording, output_folder, verbose=verbose)
---> 86     SorterClass.run_from_folder(output_folder, raise_error, verbose)
     87     if with_output:
     88         sorting = SorterClass.get_result_from_folder(output_folder)

~/miniconda3/envs/si/lib/python3.8/site-packages/spikeinterface-0.92.1.dev0-py3.8.egg/spikeinterface/sorters/basesorter.py in run_from_folder(cls, output_folder, raise_error, verbose)
    234         if has_error and raise_error:
    235             print(log['error_trace'])
--> 236             raise SpikeSortingError(
    237                 f"Spike sorting failed. You can inspect the runtime trace in {output_folder}/spikeinterface_log.json")
    238 

SpikeSortingError: Spike sorting failed. You can inspect the runtime trace in results_KS3/spikeinterface_log.json

alejoe91, 2021-12-19:
Hi, is it possible that you are missing some Matlab toolboxes? Are you able to run the ks3 example from the repo?

samuelgarcia, 2022-01-03:
To debug this you can truy to run from matlab the script `kilosort3_master.m` which is generated in your folder.

jjtukker, 2022-01-07:
Thanks for your input!
I have a campus license (R2021a) with pretty much all toolboxes, so I don't think that can be an issue. 

kilosort3_master.m actually appears to run fine, it gets beyond the point that the tutorial gets stuck at (which appears to be extract_spikes). 

Turns out the issue is related to CUDA:
spikedetector3.cu was in the matlab path but there was no spikedetector3.mexa64 (note the error says " undefined function 'spikedetector3' "). Something must have gone wrong when I last ran mexGPUall. Redoing this resolved the issue! Unfortunately it does not quite run to completion, though. Not sure if this requires a new issue. The new error is at the export to phy stage:

Saving results to Phy

Saving results to Phy

--------------------------------------------------------------------------------Error using -

Invalid type of input arguments (should be uint64)Error using -

Invalid type of input arguments (should be uint64)
Error running kilosort3
Traceback (most recent call last):
  File "/home/john/miniconda3/envs/si/lib/python3.8/site-packages/spikeinterface-0.92.1.dev0-py3.8.egg/spikeinterface/sorters/basesorter.py", line 200, in run_from_folder
    SorterClass._run_from_folder(output_folder, sorter_params, verbose)
  File "/home/john/miniconda3/envs/si/lib/python3.8/site-packages/spikeinterface-0.92.1.dev0-py3.8.egg/spikeinterface/sorters/kilosortbase.py", line 42, in _run_from_folder
    raise Exception(f'{cls.sorter_name} returned a non-zero exit code')
Exception: kilosort3 returned a non-zero exit code

---------------------------------------------------------------------------
SpikeSortingError                         Traceback (most recent call last)
/tmp/ipykernel_4086238/2606529102.py in <module>
----> 1 sorting_KS3 = ss.run_kilosort3(recording_saved, 
      2                               output_folder='results_KS3',
      3                               verbose=True)

~/miniconda3/envs/si/lib/python3.8/site-packages/spikeinterface-0.92.1.dev0-py3.8.egg/spikeinterface/sorters/runsorter.py in run_kilosort3(*args, **kwargs)
    368 
    369 def run_kilosort3(*args, **kwargs):
--> 370     return run_sorter('kilosort3', *args, **kwargs)
    371 
    372 

~/miniconda3/envs/si/lib/python3.8/site-packages/spikeinterface-0.92.1.dev0-py3.8.egg/spikeinterface/sorters/runsorter.py in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, with_output, **sorter_params)
     55 
     56     if docker_image is None:
---> 57         sorting = run_sorter_local(sorter_name, recording, output_folder=output_folder,
     58                                    remove_existing_folder=remove_existing_folder,
     59                                    delete_output_folder=delete_output_folder,

~/miniconda3/envs/si/lib/python3.8/site-packages/spikeinterface-0.92.1.dev0-py3.8.egg/spikeinterface/sorters/runsorter.py in run_sorter_local(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, **sorter_params)
     84         recording, output_folder, sorter_params, verbose)
     85     SorterClass.setup_recording(recording, output_folder, verbose=verbose)
---> 86     SorterClass.run_from_folder(output_folder, raise_error, verbose)
     87     if with_output:
     88         sorting = SorterClass.get_result_from_folder(output_folder)

~/miniconda3/envs/si/lib/python3.8/site-packages/spikeinterface-0.92.1.dev0-py3.8.egg/spikeinterface/sorters/basesorter.py in run_from_folder(cls, output_folder, raise_error, verbose)
    234         if has_error and raise_error:
    235             print(log['error_trace'])
--> 236             raise SpikeSortingError(
    237                 f"Spike sorting failed. You can inspect the runtime trace in {output_folder}/spikeinterface_log.json")
    238 

SpikeSortingError: Spike sorting failed. You can inspect the runtime trace in results_KS3/spikeinterface_log.json

alejoe91, 2022-01-13:
can you try to re-install Kilosort3?

jjtukker, 2022-01-19:
thanks for your suggestion!
I re-installed kilosort from the tutorial jupyter notebook.
This seemed to work (after I removed the previous Kilosort directory):
!git clone https://github.com/MouseLand/Kilosort
Cloning into 'Kilosort'...
remote: Enumerating objects: 3189, done.
remote: Counting objects: 100% (144/144), done.
remote: Compressing objects: 100% (100/100), done.
remote: Total 3189 (delta 86), reused 81 (delta 44), pack-reused 3045
Receiving objects: 100% (3189/3189), 134.93 MiB | 8.33 MiB/s, done.
Resolving deltas: 100% (2120/2120), done.

Then I ran mexGPUall in the CUDA directory, this also seemed to work.
Unfortunately I still get the same output at the "save to phy" stage, as shown above. I guess this error is coming from CUDA (?)
"Error using - Invalid type of input arguments (should be uint64)"

mexGPUall did generate some warnings regarding nvcc, maybe this gives some insight?
>> mexGPUall
Building with 'nvcc'.
nvcc warning : The 'compute_35', 'compute_37', 'compute_50', 'sm_35', 'sm_37' and 'sm_50' architectures are deprecated, and may be removed in a future release (Use -Wno-deprecated-gpu-targets to suppress warning).
/home/john/spikesorting/spikeinterface/tutorials/Kilosort/CUDA/spikedetector3.cu(65): warning: variable "C0" was declared but never referenced
/home/john/spikesorting/spikeinterface/tutorials/Kilosort/CUDA/spikedetector3.cu(159): warning: variable "NchanUp" was set but never used
/home/john/spikesorting/spikeinterface/tutorials/Kilosort/CUDA/spikedetector3.cu(161): warning: variable "d2" was declared but never referenced
/home/john/spikesorting/spikeinterface/tutorials/Kilosort/CUDA/spikedetector3.cu(230): warning: variable "nt0" was set but never used
/home/john/spikesorting/spikeinterface/tutorials/Kilosort/CUDA/spikedetector3.cu(20): warning: variable "nt0max" was declared but never referenced
/home/john/spikesorting/spikeinterface/tutorials/Kilosort/CUDA/spikedetector3.cu(20): warning: variable "NchanMax" was declared but never referenced
/home/john/spikesorting/spikeinterface/tutorials/Kilosort/CUDA/spikedetector3.cu(65): warning: variable "C0" was declared but never referenced
/home/john/spikesorting/spikeinterface/tutorials/Kilosort/CUDA/spikedetector3.cu(159): warning: variable "NchanUp" was set but never used
/home/john/spikesorting/spikeinterface/tutorials/Kilosort/CUDA/spikedetector3.cu(161): warning: variable "d2" was declared but never referenced
/home/john/spikesorting/spikeinterface/tutorials/Kilosort/CUDA/spikedetector3.cu(230): warning: variable "nt0" was set but never used
/home/john/spikesorting/spikeinterface/tutorials/Kilosort/CUDA/spikedetector3.cu(20): warning: variable "nt0max" was declared but never referenced
/home/john/spikesorting/spikeinterface/tutorials/Kilosort/CUDA/spikedetector3.cu(20): warning: variable "NchanMax" was declared but never referenced
/home/john/spikesorting/spikeinterface/tutorials/Kilosort/CUDA/spikedetector3.cu(65): warning: variable "C0" was declared but never referenced
/home/john/spikesorting/spikeinterface/tutorials/Kilosort/CUDA/spikedetector3.cu(159): warning: variable "NchanUp" was set but never used
/home/john/spikesorting/spikeinterface/tutorials/Kilosort/CUDA/spikedetector3.cu(161): warning: variable "d2" was declared but never referenced
/home/john/spikesorting/spikeinterface/tutorials/Kilosort/CUDA/spikedetector3.cu(230): warning: variable "nt0" was set but never used
/home/john/spikesorting/spikeinterface/tutorials/Kilosort/CUDA/spikedetector3.cu(20): warning: variable "nt0max" was declared but never referenced
/home/john/spikesorting/spikeinterface/tutorials/Kilosort/CUDA/spikedetector3.cu(20): warning: variable "NchanMax" was declared but never referenced
/home/john/spikesorting/spikeinterface/tutorials/Kilosort/CUDA/spikedetector3.cu(65): warning: variable "C0" was declared but never referenced
/home/john/spikesorting/spikeinterface/tutorials/Kilosort/CUDA/spikedetector3.cu(159): warning: variable "NchanUp" was set but never used
/home/john/spikesorting/spikeinterface/tutorials/Kilosort/CUDA/spikedetector3.cu(161): warning: variable "d2" was declared but never referenced
/home/john/spikesorting/spikeinterface/tutorials/Kilosort/CUDA/spikedetector3.cu(230): warning: variable "nt0" was set but never used
/home/john/spikesorting/spikeinterface/tutorials/Kilosort/CUDA/spikedetector3.cu(20): warning: variable "nt0max" was declared but never referenced
/home/john/spikesorting/spikeinterface/tutorials/Kilosort/CUDA/spikedetector3.cu(20): warning: variable "NchanMax" was declared but never referenced
/home/john/spikesorting/spikeinterface/tutorials/Kilosort/CUDA/spikedetector3.cu(65): warning: variable "C0" was declared but never referenced
/home/john/spikesorting/spikeinterface/tutorials/Kilosort/CUDA/spikedetector3.cu(159): warning: variable "NchanUp" was set but never used
/home/john/spikesorting/spikeinterface/tutorials/Kilosort/CUDA/spikedetector3.cu(161): warning: variable "d2" was declared but never referenced
/home/john/spikesorting/spikeinterface/tutorials/Kilosort/CUDA/spikedetector3.cu(230): warning: variable "nt0" was set but never used
/home/john/spikesorting/spikeinterface/tutorials/Kilosort/CUDA/spikedetector3.cu(20): warning: variable "nt0max" was declared but never referenced
/home/john/spikesorting/spikeinterface/tutorials/Kilosort/CUDA/spikedetector3.cu(20): warning: variable "NchanMax" was declared but never referenced

[... cut ...]
MEX completed successfully.
Building with 'nvcc'.
nvcc warning : The 'compute_35', 'compute_37', 'compute_50', 'sm_35', 'sm_37' and 'sm_50' architectures are deprecated, and may be removed in a future release (Use -Wno-deprecated-gpu-targets to suppress warning).

nvcc warning : The 'compute_35', 'compute_37', 'compute_50', 'sm_35', 'sm_37' and 'sm_50' architectures are deprecated, and may be removed in a future release (Use -Wno-deprecated-gpu-targets to suppress warning).

MEX completed successfully.

alejoe91, 2022-01-20:
@jjtukker I would open an issue on the Kilosort github repo. I'm sure they can help more than use for this specific installation issue!

jjtukker, 2022-01-20:
unfortunately the kilosort repo does not seem to be responding to any issues, at least none that I could see going back to August.. I posted this issue there anyway (https://github.com/MouseLand/Kilosort/issues/469 ), but I'm afraid this means kilosort is just off the table for now. Thanks anyway for the comment!

closed 2022-04-01 by alejoe91