SpikeInterface/spikeinterface #421: compute_amplitudes_cutoff(we) returns 0.5 for all units in recording

Yizhar-Lab, 2021-12-19:
Hi,
Some of these units actually have nice bell-shape peak distributions like this:
[image]
Here's how I get the waveform extractor:
```
we = si.extract_waveforms(recording, sorting, wf_dir,
                load_if_exists=False, overwrite=True, ms_before=0.25, 
                ms_after=0.75, max_spikes_per_unit=1e5,n_jobs=1, chunk_size=30000)
```
I can also provide the raw data files, if necessary to solve this. They're ~5GB.
Thanks!

alejoe91, 2021-12-20:
Hi @Yizhar-Lab 

I just tried with a test dataset and cutoffs seem to be computed correctly on my side:
[image]

Note that in case the cutoff is higher than 0.5 it's set to 0.5 (see [here](https://allensdk.readthedocs.io/en/latest/_static/examples/nb/ecephys_quality_metrics.html#Amplitude-cutoff)), so that explains the 0.5 values. In your case, however, it doesn't seem to be the case though! It would be great if you can share the raw data (recording + sorting output).

Cheers
Alessio

Yizhar-Lab, 2021-12-20:
Thanks @alejoe91!
Here's the [binary recording dat file](https://www.dropbox.com/s/v5y3uy9kozpwaqu/all_chans.dat?dl=0), the [manual sorting text file](https://www.dropbox.com/s/6ml2ukmefg0vq7n/Asaf_all_chans.txt?dl=0), the [dumped waveform extractor](https://www.dropbox.com/s/95oszaxh3qcnfaz/wfs_dump.pickle?dl=0), which includes "its version" of the recording and sorting,  in case that helps. (you'll have to ask my approval to access the files)

Yizhar-Lab, 2021-12-20:
P.S. I'm aware that 0.5 is the maximal value this metric can take and I've gotten realistic values from this function before. I guess there's something wrong with the extracted waveforms in this recording specifically, but I'm not sure what and why. Looking into the extracted waveforms with `wfs = we.get_waveforms(unit_id)`, gives me an `n_spikes`-by-`n_wf_samples`-by-`n_channels `array. Perhaps the problem is that the metric is calculated based on the wrong channel? How does it know which channels the units correspond to when the sorting is manual and extracted using `se.NumpySorting.from_times_labels()`?

alejoe91, 2021-12-20:
The best channel is computed based on the amplitude of the template (the average waveforms), which by default looks for the channel with the most negative amplitude. Is that the case for you recording? 

How did you get the distribution showd above?

Yizhar-Lab, 2021-12-20:
Actually my recording has positive spikes. Can I specify this for a `NumpySorting`?

I got the distribution above just by calculating it manually - getting the peaks of the spikes from the correct (most positive amp) channel and taking their histogram.

alejoe91, 2021-12-20:
Ah ok that might explain the mismatch. The easiest thing would be to transform the input data and invert the signal. Currently the [`transform` function](https://github.com/SpikeInterface/spiketoolkit/blob/master/spiketoolkit/preprocessing/transform.py) is not ported yet to the new version. Alternatively, you should be able to pass the argument `peak_sign` to the compute metrics function, but I'm noticing now that this is not propagated everywhere yet. Will make a PR on this second option.

alejoe91, 2021-12-20:
Actually, calling `compute_quality_metrics` with the extra arg: `peak_sign="pos"` should fix it for amplitude cutoffs. Can you try?

Yizhar-Lab, 2021-12-20:
so `metrics = st.compute_quality_metrics(we, waveform_principal_component=pc, peak_sign="pos")`
gives `*** TypeError: compute_isi_violations() got an unexpected keyword argument 'peak_sign'`
while `amp_cutoffs = st.compute_amplitudes_cutoff(we, peak_sign="pos")` works and even returns one value (out of 16) that's smaller than 0.5, but the unit with the peak histogram above still incorrectly gets 0.5...

Yizhar-Lab, 2021-12-20:
I guess I could try inverting the data, but it may cause problems with other analysis pipelines expecting these data to be positive... Anyway, if you do end up adding the `peak_sign` option, I'd appreciate it if you could mention it hear so I could try it out.
Thanks a lot again!

alejoe91, 2021-12-20:
Ok, thanks for trying. I'll look at it in the next few days! I'll write here when I make a PR

alejoe91, 2022-01-07:
We are working on a fix here:https://github.com/SpikeInterface/spikeinterface/pull/430

closed 2022-01-13 by samuelgarcia

alejoe91, 2022-01-13:
@Yizhar-Lab fixed via https://github.com/SpikeInterface/spikeinterface/pull/449

Basically we have to make the amplitudes negative before computing the amplitude cutoff