SpikeInterface/spikeinterface #424: concatenating times of recordings

khl02007, 2021-12-31:
@alejoe91 right now a concatenated recording doesn't concatenate the times of the constituent recordings. from the [docstring](https://github.com/SpikeInterface/spikeinterface/blob/e3c694115f512267634b2037d46e8e02812f5ccd/spikeinterface/core/segmentutils.py#L85) this seems like an explicit design decision. is there a reason why? it would be very useful for us to be able to concatenate the recordings as well as the times.

alejoe91, 2021-12-31:
@khl02007 indeed this was a design choice (after lots of discussion with @samuelgarcia). We'd be happy to discuss if it makes sense to relax this choice or find other solutions. Let's organize a call in the new year :)

khl02007, 2022-01-01:
@alejoe91 sure, let me know. my use case is that I have a long recording with timestamps, and I want to select disjoint intervals within the recording and sort them together. I thought I would frame_slice them and concatenate, but I also want the same to happen to the times. Should I just slice the timestamps manually and use `set_times` to add it to the concatenated recording? doctstring says using `set_times` is not recommended?

samuelgarcia, 2022-01-03:
I think we can relax this behavior with one option  and also good time checker functions across segment to not allow overlapping time vector or none increasing time across segment.

The set_times is not recommended for **spike sorting** because it is a super heavy attribute to recording pobject that slow down a lot the multi processing stuff because the full time vector is copied across process.
No sorter handle the time internally.
For data management, of course, after and before spike sorting `set_times` can be very usefull. Before export to nwb for instance or to get the real time of spikes (which is not handle in sorting object at the moment).

khl02007, 2022-02-03:
@alejoe91 I think this was resolved with https://github.com/SpikeInterface/spikeinterface/pull/489. feel free to close

closed 2022-02-03 by alejoe91