SpikeInterface/spikeinterface #435: Mismatch in number of spikes and spike amplitudes

guitchounts, 2022-01-07:
When sorting a short snippet of a recording, `export_report` fails because the number of spikes is always larger (by one) than the number of amplitudes. Example code:

```
data = se.read_openephys(base_path)
probe = read_prb(probe_path).probes[0]
recording = data.set_probe(probe)
recording = st.preprocessing.notch_filter(recording,freq=60)
recording_f = st.preprocessing.bandpass_filter(recording, freq_min=300, freq_max=6000)
recording_cmr = st.preprocessing.common_reference(recording_f, reference='global',  operator='median')

# shorten the recording for testing
recording_cmr = recording_cmr.frame_slice(start_frame=0*fs, end_frame=20*fs)
sorting_MS4 = ss.run_sorter('mountainsort4',recording_cmr, output_folder='%s/tmp_MS4' % base_path, **ms4_params)

we = si.extract_waveforms(recording_cmr, sorting_MS4,  '%s/tmp_MS4/waveforms' % base_path, ms_before=1, ms_after=1, load_if_exists=False, n_jobs=8, total_memory='1G')

amplitudes = st.get_spike_amplitudes(we, peak_sign='neg', outputs='concatenated')
```

For example, `len(amplitudes[0])` is 50602 while  `sorting_MS4.get_all_spike_trains()[0][0].shape` is 50603. As a result, `export_report(we, '%s/tmp_MS4/report' % base_path,remove_if_exists=True)` fails. This is true for several different recording sessions. Any idea why this is happening?

alejoe91, 2022-01-08:
Hi, thanks for the issue. I think that the problem arises when a spike is at the edge of the recording. @samuelgarcia can you confirm? In any case, we have to fix this soon

guitchounts, 2022-01-09:
Hi Alessio, thanks for checking on this. I can confirm that it does work for different durations of the same data, so you're probably right about there being spikes at the very edge.

samuelgarcia, 2022-01-10:
Hi Grigori.
Thanks for the feed back.
I will have a look, it must a border effect.
I guess that if you change the chunk_size in `amplitudes = st.get_spike_amplitudes(we, peak_sign='neg', outputs='concatenated')` then the bug must have various effect no ?
I will try to fix it soon.

samuelgarcia, 2022-01-10:
Other question : could you help me a do `print(peak_shifts)` line 88 in spike_amplitudes.py ?

Or easier give me the data rec+sorting) that trigger the bug.

guitchounts, 2022-01-11:
Hi @samuelgarcia thanks for the new code. I tried your branch and it seems to have fixed the issue: same number of spikes in `amplitudes` and `sorting_MS4.get_all_spike_trains()[0][0].shape` and `export_report` now works. Although for whatever reason `export_report` took about 25 minutes for a 20-second recording snippet, but that's a different issue

samuelgarcia, 2022-01-11:
Hi Grigori.
Can you confirm that the very long time is due to amplitude ?

samuelgarcia, 2022-01-11:
For me `compute_spikes_amplitudes` is not so slow:
 1800.000s 32 channel 15 units with n_jobs=4 it take me 18s to extract ampltiudes on a very old computer.

Could you all steps one by one to track what is slow ?
Like this
```python
we = extract_waveforms(..., n_jobs=XXX, progress_bar=True, chunk_size=30000)
compute_principal_components(we, ..., n_jobs=XXX, progress_bar=True, chunk_size=30000)
compute_quality_metrics(we, ......, n_jobs=XXX, progress_bar=True, chunk_size=30000)
compute_spikes_amplitudes(we, ......, n_jobs=XXX, progress_bar=True, chunk_size=30000)
```

before 
`export_report`

guitchounts, 2022-01-11:
Hi Samuel, 

Some stats: the recording snippet is 20 seconds long, from a 64-channel array. I'm running this on an 8-core node with 32GB RAM. Here's what I got:

`extract_waveforms finished in 10.83 seconds`
`compute_principal_components finished in 25.54 seconds`
`compute_quality_metrics finished in 389.15 seconds`
`compute_spike_amplitudes finished in 1.19 seconds`

By the way, `compute_principal_components` and `compute_quality_metrics` don't accept `n_jobs` or `chunk_size` (at least in 0.92.1.dev0). This probably has something to do with it?

samuelgarcia, 2022-01-11:
Yes you are right job_kwargs are only for `compute_spike_amplitudes`.
parralell compute for other not done yet.

So ampltiudes is super fast!!

The slow come from quality_metrics. 
How many units ?
Which `metric_names` do you compute ?

closed 2022-01-11 by samuelgarcia

guitchounts, 2022-01-11:
This has 132 units. QC computes `num_spikes`,	`firing_rate`,	`presence_ratio`,	`snr`,	`isi_violations_rate`,	`isi_violations_count`,	`amplitude_cutoff`,	`isolation_distance`,	`l_ratio`,	`d_prime`

PR #439 (2022-01-11): enable back compute_spike_amplitudes with n_jobs >= 2
* enable back compute_spike_amplitudes with n_jobs >= 2
* try fix #435 with differents shift approach
* avoid loop and use numpy vector