SpikeInterface/spikeinterface #453: Add timeout to klusta sorter execution

exarchou, 2022-01-14:
I am using the spikeinterface to sort Open Ephys Recordings. Among different sorters I also use klusta sorter with the command "**_ss.run_klusta()_**". Due to the content of the data, I often observe that klusta sorter requires too long time to execute. I would like to limit its execution time to a specific period, which if exceeded, klusta will be terminated and return the neurons it has identified so far. 
For this reason, I noticed that the execution of the sorter is based on a **_ShellScript_** which is defined in the path: 

_“/home/username/anaconda3/envs/spiketutorial/lib/python3.6/site-packages/spikesorters/utils/shellscript.py”_

The script is actually executed via the “**_wait()_**” function, which is defined inside it and it also has the argument **_timeout_** which is predefined to **_None_**. I tried to modify this argument when calling the script, in order to stop the execution after a specific time period but I did not manage to succeed it. 

Could someone explain me what is the meaning of this argument? is there any way to limit the execution time of klusta sorter?

alejoe91, 2022-01-17:
Hi @exarchou 

Unfortunately, that is not possible..
Also we recommend upgrading the the new API (spikeinterface>=0.90), which has a lot of improvements and bug fixes.

Cheers
Alessio

samuelgarcia, 2022-01-17:
And a cool GUI! https://github.com/SpikeInterface/spikeinterface-gui

closed 2022-04-01 by alejoe91