SpikeInterface/spikeinterface #455: Generic sorter output

JuliaSprenger, 2022-01-14:
From what I have seen when running sorters in containers, the output format of a `run_sorter` call is specific to the sorter used. Each format can be loaded by the corresponding sorter specific extractor for further usage. At least some of those extractors require the original sorter to be installed, e.g. in the case of tridesclous. When the sorting is performed locally this approach works fine as the sorting software anyway needs to be installed on the system.

However, in the case of containerized sorter the dependency on the sorter specific extractor implies that the sorter still needs to be installed locally, even though the sorting was performed in dedicated container.

I think it might be good if the sorter output would be independent of the sorter, to not be required to have two installations of the sorter (locally and containerized). What do you think about this and which format could be used for this generic sorter output?

alejoe91, 2022-01-14:
Hi @JuliaSprenger 

I think this was changed with the last refactor. In my opinion, we should add the following lines [here](https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/sorters/runsorter.py#L256):
```
# add in the imports of the pyscript
from spikeinterface import load_extractor, NpzSortingExtractor
from spikeinterface.sorters import run_sorter_local, sorter_dict

run_sorter_local('{sorter_name}', recording, output_folder=output_folder,
            remove_existing_folder={remove_existing_folder}, delete_output_folder=False,
            verbose={verbose}, raise_error={raise_error}, **sorter_params)
SorterClass = sorter_dict[sorter_name]
sorting = SorterClass.get_result_from_folder(output_folder)
NpzSortingExtractor.write_sorting(sorting, output_folder / "sorting.npz")
```

Then [here](https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/sorters/runsorter.py#L351) the loading would be would be:
```
sorting = NpzSortingExtractor(output_folder / "sorting.npz")
```

What do you think?

samuelgarcia, 2022-01-17:
Hi Julia,
yes you are right. This is something that I had in mind since long time.
tridesclous is the only sorter that need to be install for read the output.  All other sorters don't dpeend on the package but on neo or directly si for reading the output.
Since tdc was easy to install in any case it was not a problem.
Now TDC is pain because of hdbscan/numba/numpy and so it became a problem.
I will try something like Alessio proposed with some option when necessary.
But I think it is not good that the NPZ became be "the" format after sorting in any case.
The other way is to implement a reader for TDC that do not depend on TDC which is best I think.

JuliaSprenger, 2022-01-17:
@samuelgarcia  Ok, then I leave this to you to decide which solution you prefer and don't include a fix for this in https://github.com/SpikeInterface/spikeinterface/pull/440

alejoe91, 2022-08-01:
implemented backup save to npz

closed 2022-08-01 by alejoe91