SpikeInterface/spikeinterface #459: `UnitsMergeSorting`

khl02007, 2022-01-15:
`UnitsSelectionSorting` is very useful because it allows the user to select certain units based on e.g. quality metrics. I was wondering if it would be a good idea to have a class like this but for merging; e.g. merges a set of pairs of units specified by the user and returns the modified sorting object. Given that discarding bad/noise units and merging overspilt units are probably the two things that should be done during curation (we probably should stay away from splitting), this feature might be useful. Thoughts?

alejoe91, 2022-01-15:
Related to https://github.com/SpikeInterface/spikeinterface/issues/254

samuelgarcia, 2022-01-17:
Hi Kyu.
Yes this is something we have in mind. Merge and also split.
After this we wanted to write a class (manual cleaner) that keep history of select(aka trash)/split/merge actions.
And we could extend the spikeinterface-gui with that in mind.
It is a quite big piece of work.
I now focus on sorting_component and drift correction after this we will maybe go for that.

alejoe91, 2022-04-01:
Temporary implementation [here](https://github.com/LorenFrankLab/spyglass/blob/master/src/nwb_datajoint/spikesorting/merged_sorting_extractor.py)

Full curation object coming soon. Closing for now

closed 2022-04-01 by alejoe91