SpikeInterface/spikeinterface #460: maxwell h5 filter insider docker

ChrisCurrin, 2022-01-15:
I have maxwell-compressed .h5 data files that need their "filter" (as HDF5 consortium call them) to be read.

Locally, I set the `HDF5_PLUGIN_PATH` environment variable to the folder that contains their 'filter'.

Unfortunately, this environment variable is lost when creating a docker container.

Suggestions:

1. it may be useful to allow `run_sorter_docker` to have an environment keyword argument? This may complicate the higher-order calls to `run_sorter(s)` but allows this sort of flexibility.
2. Detect if HDF5_PLUGIN_PATH is specified in the local environment and propagate this to the docker instance (with appropriate binding of value)
e.g. 
```python
# spikeinterface.sorters.runsorter.run_sorter_docker line 192
hdf5_plugin_path = os.getenv("HDF5_PLUGIN_PATH")
if hdf5_plugin_path:
    # currently "environment" is not set elsewhere so this would be the only key definition. 
    # if this changes in the future, need to update the environment dict
    extra_kwargs["environment"] = {"HDF5_PLUGIN_PATH": hdf5_plugin_path}
    volumes[hdf5_plugin_path] = {'bind': hdf5_plugin_path, 'mode': 'ro'}
```
3. There is/was an `auto_maxwell_compression_<something or other method>` ... that should download the filter automatically upon maxwell read error, but this didn't work for me, unfortunately. Not sure what's up there. 

relatedly, but perhaps a secondary minor issue, is that copying data for spykingcircus to a numpy file in the sorting output path means the process takes longer and uses more space. Having an option to delete this file afterwards would be nice (instead of the whole output folder being the only delete option).

Much appreciated gents

samuelgarcia, 2022-01-17:
Hi Christopher,
yes this kinf of detail is a pain when using docker.
Globally this plugin stuff of maxwell is a pain.

Normally neo have a way to auto install the plugin. See https://github.com/NeuralEnsemble/python-neo/blob/master/neo/rawio/maxwellrawio.py#L222
Maybe you could install in the docker image before running the sorter.

But, yes, definitivelly we need to had more option for binding in containers.
For instance we also need to bing some volume for trained NN for yass.

What I would do for now is to save the recording to binary with this
```rec2 = rec.save('/other/folder/for/recording')``` even if it take lot of space.
You can delete the folder after the docker run.
The `save()` method of spikeinterface is fast and can be used in parralel.

I leave this PR open.

alejoe91, 2022-04-01:
@ChrisCurrin for this issue we recommend to create a Docker image based on the sorter ones and to install the Maxwell HDF plugin on top :)

closed 2022-04-01 by alejoe91